The-OpenROAD-Project/OpenLane #2029: Increasing DRT_OPT_ITERS Above Default Crashes Detailed Routing

Peter-Herrmann, 2023-10-27:
### Description

If I increase the value of `DRT_OPT_ITERS` to any value greater than the default (64), the flow fails saying the wrong number of arguments were passed to droute.tcl.
```
Error: droute.tcl, 38 Wrong number of arguments :utl::warn tool id msg  argument 2
child process exited abnormally
```
If I change the `DRT_OPT_ITERS` back to 64 (the default) or remove it, the flow continues normally.

### Expected Behavior

The flow should not crash when changing this value. If there is a max value intended it should be documented and error feedback should be more clear. (Also the max should be greater than 64 for this config, violations are still reducing on my design as late as the 57th iteration)

### Environment report

```shell
Kernel: Linux v6.2.0-35-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.6 (OK)
OpenLane Git Version: d054702b2cce04761cc2bc598f6b95c9d8ca7c6c
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: FAILED
/home/prherrma/runners/r4/_work/caravel_user_project/openlane/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/prherrma/runners/r4/_work/caravel_user_project/openlane/dependencies/verify_versions.py", line 76, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/prherrma/runners/r4/_work/caravel_user_project/openlane/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

d054702 2023-07-19T16:09:15+03:00 remove `unset_propagated_clock` (#1908) - passant5 -  (grafted, HEAD, tag: 2023.07.19)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.partaa.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/13192021/issue_reproducible.partaa.tar.gz)
[issue_reproducible.partab.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/13192024/issue_reproducible.partab.tar.gz)
[issue_reproducible.partac.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/13192026/issue_reproducible.partac.tar.gz)

### Relevant log output

```shell
[STEP 26]
[INFO]: Running Detailed Routing (log: ../home/prherrma/runners/r0/_work/tapeout-ci-2311/tapeout-ci-2311/openlane/user_project_wrapper/runs/23_10_26_22_18/logs/routing/26-detailed.log)...
Error: : during executing openroad script /openlane/scripts/openroad/droute.tcl
Error: : Log: ../home/prherrma/runners/r0/_work/tapeout-ci-2311/tapeout-ci-2311/openlane/user_project_wrapper/runs/23_10_26_22_18/logs/routing/26-detailed.log
Error: : Last 10 lines:
Using 1e+03 for resistance...
Using 1e-09 for time...
Using 1e+00 for voltage...
Using 1e-03 for current...
Using 1e-09 for power...
Using 1e-06 for distance...
Reading design constraints file at '/home/prherrma/runners/r0/_work/tapeout-ci-2311/tapeout-ci-2311/openlane/user_project_wrapper/runs/23_10_26_22_18/tmp/18-user_project_wrapper.sdc'…
[INFO ORD-0030] Using 20 thread(s).
Error: droute.tcl, 38 Wrong number of arguments :utl::warn tool id msg  argument 2
child process exited abnormally
```

vijayank88, 2023-10-28:
@Peter-Herrmann 
AFIAK, 64 is fixed value for number of iterations to complete detail routing.

Peter-Herrmann, 2023-10-28:
@vijayank88 
If it is fixed, then the [documentation](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/reference/configuration.md) should reflect that. 

For what its worth, though, I don't think this should be a fixed value, and if it is fixed then 64 is quite low. Later optimization passes for detailed routing take a fraction of a second and quite a few of my builds reach 0 violations after more than 60 iterations. I'm certain that other builds would have succeeded if there were more iterations, but instead I have to tweak the configuration to get it to pass in under 64 which seems arbitrary.

kareefardi, 2023-10-29:
@Peter-Herrmann The value is a range from 1 to 64. Yes the documentation should be updated. As for the value itself, it's a limit from OpenROAD. I can't tell much about it.