The-OpenROAD-Project/OpenLane #2034: Misformatted SYNTH_STRATEGY

lovisXII, 2023-11-05:
### Description

Hello,
I've setup a config.json file for my flow :
```
{
    "DESIGN_NAME": "core",
    "VERILOG_FILES": "dir::src/*.v",
    "PDK": "sky130A",
    "RUN_LINTER":0,

    "CLOCK_PERIOD": 10,
    "CLOCK_PORT": "clk",
    "CLOCK_NET": "ref::$CLOCK_PORT",

    "SYNTH_AUTONAME":1,
    "SYNTH_BIN":"yosys",
    "SYNTH_STRATEGY":"DELAY",
    "SYNTH_BUFFERING":1,

    "STD_CELL_LIBRARY":"sky130_fd_sc_hd",
    "FP_PDN_SKIPTRIM": true,
    "pdk::sky130*": {
        "FP_CORE_UTIL": 45,
        "scl::sky130_fd_sc_hd": {
            "CLOCK_PERIOD": 10
        },
        "scl::sky130_fd_sc_hdll": {
            "CLOCK_PERIOD": 10
        },
        "scl::sky130_fd_sc_hs": {
            "CLOCK_PERIOD": 8
        },
        "scl::sky130_fd_sc_ls": {
            "CLOCK_PERIOD": 10,
            "MAX_FANOUT_CONSTRAINT": 5
        },
        "scl::sky130_fd_sc_ms": {
            "CLOCK_PERIOD": 10
        }
    },
    "pdk::gf180mcu*": {
        "CLOCK_PERIOD": 24.0,
        "FP_CORE_UTIL": 40,
        "MAX_FANOUT_CONSTRAINT": 4,
        "PL_TARGET_DENSITY": 0.5
    }
}
```
When I try to enter something like "DELAY 1" in the SYNTH_STRATEGY variable, or any other numbers btw 0-4. It doesn't work, I got the following error :
```
[ERROR]: during executing yosys script /openlane/scripts/yosys/synth.tcl
[ERROR]: Log: designs/core/runs/run_10ns_v2/logs/synthesis/1-synthesis.log
[ERROR]: Last 10 lines:
 Yosys 0.34 (git sha1 4a1b5599258, gcc 8.3.1 -fPIC -Os)

[TCL: yosys -import] Command name collision: found pre-existing command `cd' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `eval' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `exec' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `read' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `trace' -> skip.
[ERROR] Misformatted SYNTH_STRATEGY ("DELAY").
[ERROR] Correct format is "DELAY|AREA 0-4|0-3".
child process exited abnormally

[ERROR]: Creating issue reproducible...
```
Did I miss any stepped ?

### Expected Behavior

I would expect it to not fail since in the documentation it is written we can chosse between DELAY/AREA 0-4/0-3

### Environment report

```shell
{
    "DESIGN_NAME": "core",
    "VERILOG_FILES": "dir::src/*.v",
    "PDK": "sky130A",
    "RUN_LINTER":0,

    "CLOCK_PERIOD": 10,
    "CLOCK_PORT": "clk",
    "CLOCK_NET": "ref::$CLOCK_PORT",

    "SYNTH_AUTONAME":1,
    "SYNTH_BIN":"yosys",
    "SYNTH_STRATEGY":"DELAY",
    "SYNTH_BUFFERING":1,

    "STD_CELL_LIBRARY":"sky130_fd_sc_hd",
    "FP_PDN_SKIPTRIM": true,
    "pdk::sky130*": {
        "FP_CORE_UTIL": 45,
        "scl::sky130_fd_sc_hd": {
            "CLOCK_PERIOD": 10
        },
        "scl::sky130_fd_sc_hdll": {
            "CLOCK_PERIOD": 10
        },
        "scl::sky130_fd_sc_hs": {
            "CLOCK_PERIOD": 8
        },
        "scl::sky130_fd_sc_ls": {
            "CLOCK_PERIOD": 10,
            "MAX_FANOUT_CONSTRAINT": 5
        },
        "scl::sky130_fd_sc_ms": {
            "CLOCK_PERIOD": 10
        }
    },
    "pdk::gf180mcu*": {
        "CLOCK_PERIOD": 24.0,
        "FP_CORE_UTIL": 40,
        "MAX_FANOUT_CONSTRAINT": 4,
        "PL_TARGET_DENSITY": 0.5
    }
}
```

### Reproduction material

.

### Relevant log output

```shell
[ERROR]: during executing yosys script /openlane/scripts/yosys/synth.tcl
[ERROR]: Log: designs/core/runs/run_10ns_v2/logs/synthesis/1-synthesis.log
[ERROR]: Last 10 lines:
 Yosys 0.34 (git sha1 4a1b5599258, gcc 8.3.1 -fPIC -Os)

[TCL: yosys -import] Command name collision: found pre-existing command `cd' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `eval' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `exec' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `read' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `trace' -> skip.
[ERROR] Misformatted SYNTH_STRATEGY ("DELAY").
[ERROR] Correct format is "DELAY|AREA 0-4|0-3".
child process exited abnormally

[ERROR]: Creating issue reproducible...
```
```

kareefardi, 2023-11-05:
@lovisXII In your example, your are setting it to `DELAY`. It should be values similar to `DELAY 0`, `DELAY 1`, `AREA 0`, etc.

lovisXII, 2023-11-05:
Well wrong copy paste my mistaek, it doesnt work with DELAY 0 but it does with just DELAY

d-m-bailey, 2023-11-05:
@lovisXII The error message and the config file don't match. Maybe you haven't saved the config file (there's an `M`:modified after the filename.)
The  `Correct format` message may be a little misleading. The correct values are
`DELAY 0-4` or `AREA 0-3`.
You can check the code [here](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/yosys/synth.tcl).

lovisXII, 2023-11-05:
Indeed you're correct my script to copy the config file in the correct directory had a mistake and the file was not copied, thanks !

closed 2023-11-05 by lovisXII