The-OpenROAD-Project/OpenLane #2035: Segmentation violation at global routing resizer timing optimizations

yunuseryilmaz18, 2023-11-08:
### Description

Hello, I'm using the gfmpw-1b tagged version of the caravel-user-project to harden my macro. I got these crash errors in the Global Routing Resizer Timing Optimizations step:

```
[STEP 20]
[INFO]: Running Global Routing Resizer Timing Optimizations (log: ../home/yunus/caravel_user_project/openlane/fpga_core/runs/23_11_07_16_58/logs/routing/20-resizer_timing.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Log: ../home/yunus/caravel_user_project/openlane/fpga_core/runs/23_11_07_16_58/logs/routing/20-resizer_timing.log
[ERROR]: Last 10 lines:
36# 0x00007F6BFADE2F1E in /lib64/libtcl8.5.so
37# Tcl_EvalEx in /lib64/libtcl8.5.so
38# Tcl_Eval in /lib64/libtcl8.5.so
39# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
40# ord::tclAppInit(Tcl_Interp*) in openroad
41# Tcl_Main in /lib64/libtcl8.5.so
42# main in openroad
43# __libc_start_main in /lib64/libc.so.6
44# 0x0000000000D3DC77 in openroad
child killed: segmentation violation
[ERROR]: Creating issue reproducible...
```

Also, issue-survey failed to recognize gf180mcu. What should I do?

### Expected Behavior

I was expecting to finish the flow without errors.

### Environment report

```shell
Kernel: Linux v6.2.0-36-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.7 (OK)
OpenLane Git Version: 7ea7a2aeef4bea5445d133c26ca2b3e9f0d0c78f
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: FAILED
/home/yunus/caravel_user_project/dependencies/openlane_src/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/yunus/caravel_user_project/dependencies/openlane_src/dependencies/verify_versions.py", line 75, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/yunus/caravel_user_project/dependencies/openlane_src/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

7ea7a2a 2023-10-15T17:01:59+03:00 Fix `FP_PIN_ORDER_CFG` being overriden (#2017) - Kareem Farid -  (grafted, HEAD, tag: 2023.10.16)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/13292742/issue_reproducible.zip)

### Relevant log output

```shell
[STEP 20]
[INFO]: Running Global Routing Resizer Timing Optimizations (log: ../home/yunus/caravel_user_project/openlane/fpga_core/runs/23_11_07_16_58/logs/routing/20-resizer_timing.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Log: ../home/yunus/caravel_user_project/openlane/fpga_core/runs/23_11_07_16_58/logs/routing/20-resizer_timing.log
[ERROR]: Last 10 lines:
36# 0x00007F6BFADE2F1E in /lib64/libtcl8.5.so
37# Tcl_EvalEx in /lib64/libtcl8.5.so
38# Tcl_Eval in /lib64/libtcl8.5.so
39# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
40# ord::tclAppInit(Tcl_Interp*) in openroad
41# Tcl_Main in /lib64/libtcl8.5.so
42# main in openroad
43# __libc_start_main in /lib64/libc.so.6
44# 0x0000000000D3DC77 in openroad
child killed: segmentation violation
[ERROR]: Creating issue reproducible...
```

donn, 2023-11-08:
No worries about the issue survey, that's fine.

I'm gonna forward this issue to the OpenROAD team to see if they may have a workaround.

closed 2023-11-19 by donn

donn, 2023-11-30:
Would you kindly file a new issue? @yunuseryilmaz18

PR #2036 (2023-11-19): Update OpenROAD
~ `openroad_app` -> `6f9b2bb` to match efabless/openlane2#255

---

Resolves #2035