The-OpenROAD-Project/OpenLane #2040: Magic DRC check reporting 0 errors in Openlane and MPW local precheck

ellen-wood, 2023-11-10:
### Description

Our `$PDK` is set to Sky130B. 

Our GDS layout in Magic had 1246 DRC errors which we hadn't noticed, due to not setting the DRC style to 'full'. 

We included this same GDS into OpenLane to build user_project_wrapper. 
`[Step 30] Magic DRC` showed 0 DRC violations in the flow. 

[image]

However the `drc.rpt` and `30-drc.log` had picked up on the errors, but had not reported them: 

[image]
[30-drc.log](https://github.com/The-OpenROAD-Project/OpenLane/files/13320142/30-drc.log)

The `drc.tr` was empty (0 bytes), no data at all when trying to open it. 

We did not inspect the DRC reports so continued on to local pre-check. This also showed 0 Magic DRC errors: 

[image]

However the log for this step shows that the Magic DRC check did not run correctly:

[magic_drc_check.log](https://github.com/The-OpenROAD-Project/OpenLane/files/13320180/magic_drc_check.log)
 

We only found out about the DRC errors because another user who is helping integrate the designs was using the Sky130A PDK, and all of the errors from our GDS showed up on his pre-check. 

[image]

### Expected Behavior

I would have expected both the Openlane Magic DRC check and the local pre-check to have reported the DRCs (or at least flag that the Magic DRC check was unable to run correctly). 

It was our fault for not knowing about the two DRC decks in Magic and not checking the Full set, however if either of the Magic DRC checks had flagged up any errors, we would have known about the presence of these errors weeks ago! 

### Environment report

```shell
Kernel: Linux v5.8.1-050801-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: d054702b2cce04761cc2bc598f6b95c9d8ca7c6c
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

d054702 2023-07-19T16:09:15+03:00 remove `unset_propagated_clock` (#1908) - passant5 -  (grafted, HEAD, tag: 2023.07.19)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

Upload is not supporting file type of .part, however we have changed the GDS to remedy the errors 

### Relevant log output

```shell
Magic 8.3 revision 420 - Compiled on Sat Nov  4 21:03:17 UTC 2023.
Starting magic under Tcl interpreter
Using the terminal as the console.
Using NULL graphics device.
Processing system .magicrc file
Sourcing design .magicrc for technology sky130A ...
2 Magic internal units = 1 Lambda
Could not find file '/root/eda/pdk/sky130B/libs.tech/magic/sky130A.tech' in any of these directories:
         . $CAD_ROOT/magic/sys $CAD_ROOT/magic/sys/current
Error parsing "/root/mpw_precheck/checks/tech-files/sky130A.magicrc": couldn't read file "/root/eda/pdk/sky130B/libs.tech/magic/sky130A.tcl": no such file or directory
Bad local startup file "/root/mpw_precheck/checks/tech-files/sky130A.magicrc", continuing without.
Scaled tech values by 2 / 1 to match internal grid scaling
Loading "/root/mpw_precheck/checks/drc_checks/magic/magic_drc_check.tcl" from command line.
Don't know how to read GDS-II:
Nothing in "cifinput" section of tech file.
[INFO]: Loading user_project_wrapper

Cell user_project_wrapper couldn't be read
No such file or directory
Creating new cell
"drc(full)" is not one of the DRC styles Magic knows.
The current style is "default".
The DRC styles are: default.
No errors found.
[INFO]: COUNT: 0
[INFO]: Should be divided by 3 or 4
[INFO]: DRC Checking DONE (/root/eda/caravel_user_project/precheck_results/08_NOV_2023___10_53_00/outputs/reports/magic_drc_check.drc.report)
[INFO]: Saving mag view with DRC errors(/root/eda/caravel_user_project/precheck_results/08_NOV_2023___10_53_00/outputs/user_project_wrapper.magic.drc.mag)
[INFO]: Saved
```

d-m-bailey, 2023-11-10:
@ellen-wood Thanks for the detailed report. There has been a change made that should fix this problem. Do you still have the gds data with the errors? It would be great if you could test the change.

All you should need to do is check that your Makefile has `OPENLANE_TAG?=2023.07.19-1`, make sure it's not overwritten `env | grep OPENLANE_TAG`, run `make openlane`, and then remake the hierarchy that reads the gds with the error.

donn, 2023-11-11:
As @d-m-bailey said, this has been fixed as of 1d46ea5fb2303b571e530c56d7d91a89d9c59b16 and we've backported the changes to the MPW version of OpenLane under the tag `2023.07.19-1`.

With regard to the inconvenience, I really can only say I'm sorry. It's a mistake on my part- I forgot to remove `.read()` and then the DRC conversion script was iterating over characters instead of lines. 🙃

closed 2023-11-11 by donn