The-OpenROAD-Project/OpenLane #2042: Issue in PL resizer

KanishR1, 2023-11-11:
### Description

I have created a custom riscv processor and trying to use sky130 ram macro. While doing the normal openlane flow i am getting an error, in placement. I have attached the image here. Even after changing the MAX_TRANSITION time switch in config.json as 0.5ns, it is not getting updated. I have also attached the issue reproducible.
[image]
[RUN_2023.11.10_19.50.23.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/13326025/RUN_2023.11.10_19.50.23.zip)

donn, 2023-11-11:
Hello, please re-file this as a bug report as shown below:

[image]

donn, 2023-11-11:
Hello- please re-file this issue as a bug report as shown below:

[image]

closed 2023-11-11 by donn