The-OpenROAD-Project/OpenLane #2047: PDN pitch error

kirubakaran-g, 2023-11-18:
**While processing the openlane flow of the built-in design namely 'spm', all the forty steps in flow are complete, and the required reports and logs are generated.

But, while processing the openlane flow for a simple Half Adder Logic, I got the following error while generating the PDN:**

[STEP 6]
[INFO]: Generating PDN (log: designs/ha/runs/RUN_2023.11.18_10.26.24/logs/floorplan/6-pdn.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Log: designs/ha/runs/RUN_2023.11.18_10.26.24/logs/floorplan/6-pdn.log
[ERROR]: Last 10 lines:
[INFO]: Setting input delay to: 2.0
[WARNING STA-0337] port 'wb_clk_i' not found.
[INFO]: Setting load to: 0.033442
[INFO]: Setting clock uncertainty to: 0.25
[INFO]: Setting clock transition to: 0.15
[WARNING STA-0559] transition time can not be specified for virtual clocks.
[INFO]: Setting timing derate to: 5.0 %
[ERROR PDN-0175] Pitch 1.8400 is too small for, must be atleast 6.6000
Error: pdn_cfg.tcl, 92 PDN-0175
child process exited abnormally

[ERROR]: Creating issue reproducible...

What might be the issue? Thanks in advance.

d-mitch-bailey, 2023-11-18:
@kirubakaran-g helps if you include the reproducible, but I'm guessing that the power rail pitch has been changed to `1.84`. This is not enough space to fit all the power rails.

kirubakaran-g, 2023-11-19:
The issue_reproducible folder contains a few files. I don't know what to include! Let me include the whole run here:

root@LAPTOP-QQMU2CQ2:/openlane# ./flow.tcl -design ha
OpenLane 1e9efe9d0c4f8044d917d022a05ee1706651ed48
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

[INFO]: Using configuration in 'designs/ha/config.json'...
[INFO]: PDK Root: /root/.volare
[INFO]: Process Design Kit: sky130A
[INFO]: Standard Cell Library: sky130_fd_sc_hd
[INFO]: Optimization Standard Cell Library: sky130_fd_sc_hd
[INFO]: Run Directory: /openlane/designs/ha/runs/RUN_2023.11.19_07.23.06
[INFO]: Saving runtime environment...
[INFO]: Preparing LEF files for the nom corner...
[INFO]: Preparing LEF files for the min corner...
[INFO]: Preparing LEF files for the max corner...
[WARNING]: PNR_SDC_FILE is not set. It is recommended to write a custom SDC file for the design. Defaulting to BASE_SDC_FILE
[WARNING]: SIGNOFF_SDC_FILE is not set. It is recommended to write a custom SDC file for the design. Defaulting to BASE_SDC_FILE
[INFO]: Running linter (Verilator) (log: designs/ha/runs/RUN_2023.11.19_07.23.06/logs/synthesis/linter.log)...
[INFO]: 0 errors found by linter
[INFO]: 0 warnings found by linter
[STEP 1]
[INFO]: Running Synthesis (log: designs/ha/runs/RUN_2023.11.19_07.23.06/logs/synthesis/1-synthesis.log)...
[STEP 2]
[INFO]: Running Single-Corner Static Timing Analysis (log: designs/ha/runs/RUN_2023.11.19_07.23.06/logs/synthesis/2-sta.log)...
[STEP 3]
[INFO]: Running Initial Floorplanning (log: designs/ha/runs/RUN_2023.11.19_07.23.06/logs/floorplan/3-initial_fp.log)...
[WARNING]: Current core area is too small for the power grid settings chosen. The power grid will be scaled down.
[INFO]: Floorplanned with width 7.36 and height 5.44.
[STEP 4]
[INFO]: Running IO Placement (log: designs/ha/runs/RUN_2023.11.19_07.23.06/logs/floorplan/4-io.log)...
[STEP 5]
[INFO]: Running Tap/Decap Insertion (log: designs/ha/runs/RUN_2023.11.19_07.23.06/logs/floorplan/5-tap.log)...
[INFO]: Power planning with power {VPWR} and ground {VGND}...
[STEP 6]
[INFO]: Generating PDN (log: designs/ha/runs/RUN_2023.11.19_07.23.06/logs/floorplan/6-pdn.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Log: designs/ha/runs/RUN_2023.11.19_07.23.06/logs/floorplan/6-pdn.log
[ERROR]: Last 10 lines:
[INFO]: Setting input delay to: 2.0
[WARNING STA-0337] port 'wb_clk_i' not found.
[INFO]: Setting load to: 0.033442
[INFO]: Setting clock uncertainty to: 0.25
[INFO]: Setting clock transition to: 0.15
[WARNING STA-0559] transition time can not be specified for virtual clocks.
[INFO]: Setting timing derate to: 5.0 %
[ERROR PDN-0175] Pitch 1.8400 is too small for, must be atleast 6.6000
Error: pdn_cfg.tcl, 92 PDN-0175
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /openlane/designs/ha/runs/RUN_2023.11.19_07.23.06/issue_reproducible…
Done.
[INFO]: Reproducible packaged at 'designs/ha/runs/RUN_2023.11.19_07.23.06/issue_reproducible'.

kirubakaran-g, 2023-11-19:
These are the config.tcl file's contents:

sharvamshivam@LAPTOP-QQMU2CQ2:/mnt/e/sky130repo/openlane-git/designs/ha/runs/RUN_2023.11.19_07.50.25$ cat config.tcl
# Saved State
set ::env(BASE_SDC_FILE) "/openlane/scripts/base.sdc"
set ::env(BASIC_PREP_COMPLETE) "1"
set ::env(BOTTOM_MARGIN_MULT) "4"
set ::env(CARRY_SELECT_ADDER_MAP) "/root/.volare/sky130A/libs.tech/openlane/sky130_fd_sc_hd/csa_map.v"
set ::env(CELLS_LEF) "/root/.volare/sky130A/libs.ref/sky130_fd_sc_hd/lef/sky130_ef_sc_hd.lef /root/.volare/sky130A/libs.ref/sky130_fd_sc_hd/lef/sky130_fd_sc_hd.lef"
set ::env(CELLS_LEF_OPT) "/root/.volare/sky130A/libs.ref/sky130_fd_sc_hd/lef/sky130_ef_sc_hd.lef /root/.volare/sky130A/libs.ref/sky130_fd_sc_hd/lef/sky130_fd_sc_hd.lef"
set ::env(CELL_CLK_PORT) "CLK"
set ::env(CELL_PAD_EXCLUDE) "sky130_fd_sc_hd__tap* sky130_fd_sc_hd__decap* sky130_ef_sc_hd__decap* sky130_fd_sc_hd__fill*"
set ::env(CLOCK_BUFFER_FANOUT) "16"
set ::env(CLOCK_PERIOD) "10.0"
set ::env(CLOCK_WIRE_RC_LAYER) "met5"
set ::env(CONFIGS) "general.tcl checkers.tcl synthesis.tcl floorplan.tcl cts.tcl placement.tcl routing.tcl extraction.tcl"
set ::env(CORE_AREA) "5.52 10.88 12.88 16.32"
set ::env(CORE_HEIGHT) "5.44"
set ::env(CORE_WIDTH) "7.36"
set ::env(CTS_CLK_BUFFER_LIST) "sky130_fd_sc_hd__clkbuf_8 sky130_fd_sc_hd__clkbuf_4 sky130_fd_sc_hd__clkbuf_2"
set ::env(CTS_CLK_MAX_WIRE_LENGTH) "0"
set ::env(CTS_DISABLE_POST_PROCESSING) "0"
set ::env(CTS_DISTANCE_BETWEEN_BUFFERS) "0"
set ::env(CTS_MAX_CAP) "1.53169"
set ::env(CTS_MULTICORNER_LIB) "1"
set ::env(CTS_REPORT_TIMING) "1"
set ::env(CTS_ROOT_BUFFER) "sky130_fd_sc_hd__clkbuf_16"
set ::env(CTS_SINK_CLUSTERING_MAX_DIAMETER) "50"
set ::env(CTS_SINK_CLUSTERING_SIZE) "25"
set ::env(CTS_TOLERANCE) "100"
set ::env(CURRENT_DEF) "/openlane/designs/ha/runs/RUN_2023.11.19_07.50.25/tmp/floorplan/5-tapcell.def"
set ::env(CURRENT_GUIDE) "0"
set ::env(CURRENT_INDEX) "6"
set ::env(CURRENT_LIB) "/openlane/designs/ha/runs/RUN_2023.11.19_07.50.25/results/synthesis/ha.lib"
set ::env(CURRENT_NETLIST) "/openlane/designs/ha/runs/RUN_2023.11.19_07.50.25/results/synthesis/ha.v"
set ::env(CURRENT_ODB) "/openlane/designs/ha/runs/RUN_2023.11.19_07.50.25/tmp/floorplan/5-tapcell.odb"
set ::env(CURRENT_POWERED_NETLIST) "0"
set ::env(CURRENT_SDC) "/openlane/designs/ha/runs/RUN_2023.11.19_07.50.25/tmp/floorplan/3-initial_fp.sdc"
set ::env(CURRENT_SDF) "/openlane/designs/ha/runs/RUN_2023.11.19_07.50.25/results/synthesis/ha.sdf"
set ::env(CURRENT_STEP) "floorplan"
set ::env(CVC_SCRIPTS_DIR) "/root/.volare/sky130A/libs.tech/openlane/cvc"
set ::env(DATA_WIRE_RC_LAYER) "met2"
set ::env(DECAP_CELL) "sky130_ef_sc_hd__decap_12 sky130_fd_sc_hd__decap_8 sky130_fd_sc_hd__decap_6 sky130_fd_sc_hd__decap_4 sky130_fd_sc_hd__decap_3"
set ::env(DEFAULT_SDC_FILE) "/openlane/scripts/base.sdc"
set ::env(DEF_UNITS_PER_MICRON) "1000"
set ::env(DESIGN_CONFIG) "/openlane/designs/ha/config.json"
set ::env(DESIGN_DIR) "/openlane/designs/ha"
set ::env(DESIGN_IS_CORE) "1"
set ::env(DESIGN_NAME) "ha"
set ::env(DETAILED_ROUTER) "tritonroute"
set ::env(DIE_AREA) "0.0 0.0 18.46 29.18"
set ::env(DIODE_CELL) "sky130_fd_sc_hd__diode_2"
set ::env(DIODE_CELL_PIN) "DIODE"
set ::env(DIODE_ON_PORTS) "none"
set ::env(DIODE_PADDING) "2"
set ::env(DONT_USE_CELLS) "sky130_fd_sc_hd__a2111oi_0 sky130_fd_sc_hd__a21boi_0 sky130_fd_sc_hd__and2_0 sky130_fd_sc_hd__buf_16 sky130_fd_sc_hd__clkdlybuf4s15_1 sky130_fd_sc_hd__clkdlybuf4s18_1 sky130_fd_sc_hd__fa_4 sky130_fd_sc_hd__lpflow_bleeder_1 sky130_fd_sc_hd__lpflow_clkbufkapwr_1 sky130_fd_sc_hd__lpflow_clkbufkapwr_16 sky130_fd_sc_hd__lpflow_clkbufkapwr_2 sky130_fd_sc_hd__lpflow_clkbufkapwr_4 sky130_fd_sc_hd__lpflow_clkbufkapwr_8 sky130_fd_sc_hd__lpflow_clkinvkapwr_1 sky130_fd_sc_hd__lpflow_clkinvkapwr_16 sky130_fd_sc_hd__lpflow_clkinvkapwr_2 sky130_fd_sc_hd__lpflow_clkinvkapwr_4 sky130_fd_sc_hd__lpflow_clkinvkapwr_8 sky130_fd_sc_hd__lpflow_decapkapwr_12 sky130_fd_sc_hd__lpflow_decapkapwr_3 sky130_fd_sc_hd__lpflow_decapkapwr_4 sky130_fd_sc_hd__lpflow_decapkapwr_6 sky130_fd_sc_hd__lpflow_decapkapwr_8 sky130_fd_sc_hd__lpflow_inputiso0n_1 sky130_fd_sc_hd__lpflow_inputiso0p_1 sky130_fd_sc_hd__lpflow_inputiso1n_1 sky130_fd_sc_hd__lpflow_inputiso1p_1 sky130_fd_sc_hd__lpflow_inputisolatch_1 sky130_fd_sc_hd__lpflow_isobufsrc_1 sky130_fd_sc_hd__lpflow_isobufsrc_16 sky130_fd_sc_hd__lpflow_isobufsrc_2 sky130_fd_sc_hd__lpflow_isobufsrc_4 sky130_fd_sc_hd__lpflow_isobufsrc_8 sky130_fd_sc_hd__lpflow_isobufsrckapwr_16 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_1 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_2 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_4 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_4 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_1 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_2 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_4 sky130_fd_sc_hd__mux4_4 sky130_fd_sc_hd__o21ai_0 sky130_fd_sc_hd__o311ai_0 sky130_fd_sc_hd__or2_0 sky130_fd_sc_hd__probe_p_8 sky130_fd_sc_hd__probec_p_8 sky130_fd_sc_hd__xor3_1 sky130_fd_sc_hd__xor3_2 sky130_fd_sc_hd__xor3_4 sky130_fd_sc_hd__xnor3_1 sky130_fd_sc_hd__xnor3_2 sky130_fd_sc_hd__xnor3_4  "
set ::env(DPL_CELL_PADDING) "0"
set ::env(DRC_EXCLUDE_CELL_LIST) "/root/.volare/sky130A/libs.tech/openlane/sky130_fd_sc_hd/drc_exclude.cells"
set ::env(DRC_EXCLUDE_CELL_LIST_OPT) "/root/.volare/sky130A/libs.tech/openlane/sky130_fd_sc_hd/drc_exclude.cells"
set ::env(DRT_OPT_ITERS) "64"
set ::env(FAKEDIODE_CELL) "sky130_ef_sc_hd__fakediode_2"
set ::env(FILL_CELL) "sky130_fd_sc_hd__fill*"
[... cut ...]
set ::env(synth_report_prefix) "/openlane/designs/ha/runs/RUN_2023.11.19_07.50.25/reports/synthesis/1-synthesis"
set ::env(synthesis_logs) "/openlane/designs/ha/runs/RUN_2023.11.19_07.50.25/logs/synthesis"
set ::env(synthesis_reports) "/openlane/designs/ha/runs/RUN_2023.11.19_07.50.25/reports/synthesis"
set ::env(synthesis_results) "/openlane/designs/ha/runs/RUN_2023.11.19_07.50.25/results/synthesis"
set ::env(synthesis_tmpfiles) "/openlane/designs/ha/runs/RUN_2023.11.19_07.50.25/tmp/synthesis"
set ::env(timer_start) "1700380225"

donn, 2023-11-20:
Could I kindly ask you to re-file this issue using this link? https://github.com/The-OpenROAD-Project/OpenLane/issues/new?assignees=&labels=&projects=&template=bug_report.yml The bug report template helps us address your issue faster.

Thank you.

closed 2023-11-20 by donn