The-OpenROAD-Project/OpenLane #2050: Request to Bump Yosys (downgrade) to 0.33+6 (git sha1 41b34a193)

dlmiles, 2023-11-19:
### Description

Please consider revising the resent closed PR #2027  (Bump Yosys) back a few weeks to:

Yosys 0.33+6 (git sha1 41b34a193, clang 10.0.0-4ubuntu1 -fPIC -Os)

Due to flow issues concerning:

https://github.com/YosysHQ/yosys/issues/4039

Please feel free to close this issue once you have considered the situation and made a verdict no matter the outcome, I provide this issue to advise of on-going matter.

Thanks

### Expected Behavior

yosys-abc Assertion triggered after processing a significant % of designs (estimated worse than 3%)

Assertion causes STEP1 to fail sure to abort/core-dump like crash, exit status 134

### Environment report

```shell
Linux OpenLane docker
```

### Reproduction material

See linked Yosys issue for test case and log outputs.

### Relevant log output

```shell
See linked Yosys issue for test case and log outputs.
```

kareefardi, 2023-11-19:
@dlmiles Just to confirm. Downgrading fixes the crash. In addition, using the new commands in #2051 also fixes the crash ?

dlmiles, 2023-11-19:
> @dlmiles Just to confirm. Downgrading fixes the crash. In addition, using the new commands in #2051 also fixes the crash ?

Yes that is correct

donn, 2023-11-20:
Sorry, I don't think we'll be downgrading- for OpenLane 1 it may be doable, for OpenLane 2 we have a number of plugins that rely on this newer version of Yosys, including `eqy`, `verible`, `lighter`… And we do keep the tool versions synchronized for now.

I've reported the true issue in isolation in #2052.

closed 2023-11-20 by donn