The-OpenROAD-Project/OpenLane #2058: [ERROR]: There are violations in the design after Magic DRC.

tdtu1002, 2023-11-24:
I'm a newbie and I'm having problems with my design. when running Magic DRC. Magic says I have no DRC errors but there are 56000 violations, what are the violations here? how to fix it ? thanks

[image]
[image]
[image]

d-m-bailey, 2023-11-24:
Are you using openram? openram has different rules - normal drc will flag a lot of errors. You might try temporarily replacing the sram mag file with maglef.

tdtu1002, 2023-11-25:
i follow openlane flow, and it generate this above image :(((

tdtu1002, 2023-11-30:
@dm-bailey how can i replace mag to maglef, SRAM i have gds, lef and lib file.

d-m-bailey, 2023-11-30:
For running DRC in magic, I think you load the maglef views of the sram first and then load your design and then run DRC.
maglef views may still have drc errors, but I think they're just false wide metal spacing and maybe invalid overlaps.
@RTimothyEdwards ?

RTimothyEdwards, 2023-11-30:
There are multiple ways of doing it, but it depends on whether your run DRC on a .mag file or a .gds file.  If it's a .mag file then you can either (1) load the maglef view of the SRAM or load the LEF view of the SRAM first, then load the design, which is what Mitch suggested above, or (2) load the design, then use `cellname filepath` to change the location of the SRAM to the maglef directory before expanding;  and if it's being run from GDS, then (3) load the maglef view of the SRAM first and set `gds noduplicates true` before reading the GDS.  Even after expansion, you can still replace the SRAM with `cellname filepath` followed by `flush`.

kareefardi, 2023-12-21:
@tdtu1002 can you upload a GDS or a DEF file ?

tdtu1002, 2023-12-21:
> @tdtu1002 can you upload a GDS or a DEF file ?

thanks so much everyone, i solve it by add an options to my config.tcl "MAGIC_DRC_USE_GDS" : 0,

closed 2023-12-24 by kareefardi