The-OpenROAD-Project/OpenLane #2060: Segmentation violation at placement resizer design optimizations

yunuseryilmaz18, 2023-11-30:
### Description

Hello, I'm using the gfmpw-1b tagged version of the caravel-user-project to harden my macro. I got these crash errors in the Placement Resizer Design Optimizations step:
```
[STEP 12]
[INFO]: Running Placement Resizer Design Optimizations (log: ../home/yunus/caravel_user_project/openlane/fpga_core_v2/runs/23_11_30_12_02/logs/placement/12-resizer.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer.tcl
[ERROR]: Log: ../home/yunus/caravel_user_project/openlane/fpga_core_v2/runs/23_11_30_12_02/logs/placement/12-resizer.log
[ERROR]: Last 10 lines:
41# 0x00007F01CD268F1E in /lib64/libtcl8.5.so
42# Tcl_EvalEx in /lib64/libtcl8.5.so
43# Tcl_Eval in /lib64/libtcl8.5.so
44# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
45# ord::tclAppInit(Tcl_Interp*) in openroad
46# Tcl_Main in /lib64/libtcl8.5.so
47# main in openroad
48# __libc_start_main in /lib64/libc.so.6
49# 0x0000000000D3DC77 in openroad
child killed: segmentation violation
```

### Expected Behavior

I was expecting to finish the flow without errors.

### Environment report

```shell
Kernel: Linux v6.2.0-37-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.7 (OK)
OpenLane Git Version: 7ea7a2aeef4bea5445d133c26ca2b3e9f0d0c78f
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: FAILED
/home/yunus/caravel_user_project/dependencies/openlane_src/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/yunus/caravel_user_project/dependencies/openlane_src/dependencies/verify_versions.py", line 75, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/yunus/caravel_user_project/dependencies/openlane_src/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

7ea7a2a 2023-10-15T17:01:59+03:00 Fix `FP_PIN_ORDER_CFG` being overriden (#2017) - Kareem Farid -  (grafted, HEAD, tag: 2023.10.16)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/13511946/issue_reproducible.zip)

### Relevant log output

```shell
[STEP 12]
[INFO]: Running Placement Resizer Design Optimizations (log: ../home/yunus/caravel_user_project/openlane/fpga_core_v2/runs/23_11_30_12_02/logs/placement/12-resizer.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer.tcl
[ERROR]: Log: ../home/yunus/caravel_user_project/openlane/fpga_core_v2/runs/23_11_30_12_02/logs/placement/12-resizer.log
[ERROR]: Last 10 lines:
41# 0x00007F01CD268F1E in /lib64/libtcl8.5.so
42# Tcl_EvalEx in /lib64/libtcl8.5.so
43# Tcl_Eval in /lib64/libtcl8.5.so
44# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
45# ord::tclAppInit(Tcl_Interp*) in openroad
46# Tcl_Main in /lib64/libtcl8.5.so
47# main in openroad
48# __libc_start_main in /lib64/libc.so.6
49# 0x0000000000D3DC77 in openroad
child killed: segmentation violation
```

yunuseryilmaz18, 2023-12-02:
No, I didn't set them. After seeing your comment, I set PDK variable but the result didn't change.

d-m-bailey, 2023-12-03:
Setting the variables can cause problems. I wanted you to check that they were **not** set.

d-m-bailey, 2023-12-03:
We're on `gfmpw-1c` now. Check that `make setup` runs without errors.

kareefardi, 2023-12-04:
I can confirm the issue using latest OpenROAD.

maliberty, 2023-12-04:
I started to take a look.  I'm wondering why I see:
```
read_liberty -corner Typical pdk/gf180mcuD/libs.ref/gf180mcu_fd_sc_mcu9t5v0/liberty/gf180mcu_fd_sc_mcu9t5v0__tt_025C_5v00.lib
read_liberty -corner Typical pdk/gf180mcuD/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__tt_025C_5v00.lib
```

it appears you are loading both 9t & 7t libraries which seems odd.  Is there a reason for that?

maliberty, 2023-12-04:
If I remove the 7t lib the crash is gone.  I think you are loading it only for one corner which is triggering the issue.  I'll put an error message in but I think the solution is just to drop the unneeded library.

yunuseryilmaz18, 2023-12-04:
@maliberty I'm using 9t cells for the design and if don't add a configuration variable about library, the flow gives an error about it can't find definition of the cell. What should I do to only call 9t cells?

maliberty, 2023-12-04:
@kareefardi would you explain how to do it in OL as I'm not an expert.

kareefardi, 2023-12-05:
It might be a flow error. Let me check what's happening.

kareefardi, 2023-12-05:
@yunuseryilmaz18 can you share your design input configuration ?

yunuseryilmaz18, 2023-12-05:
Yes, it did solve. Thank you!

The confusing part for me is `STD_CELL_LIBRARY_OPT` variable is `STD_CELL_LIBRARY` in default. However, in my situation it wasn't. I was defining only `STD_CELL_LIBRARY`.
[image]

The config file I used:

```
{
      "DESIGN_NAME": "grid_clb",
      "DESIGN_IS_CORE": 0,
      "VERILOG_FILES": "dir::rtl/fabric_netlists.v",
      "CLOCK_PERIOD": 300,
      "CLOCK_PORT": "top_width_0_height_0_subtile_0__pin_clk_0_",
      "STD_CELL_LIBRARY": "gf180mcu_fd_sc_mcu9t5v0",
      "RUN_LINTER": 0,
      "SYNTH_READ_BLACKBOX_LIB": 1
}
```

maliberty, 2023-12-05:
@kareefardi I think the defaults should be updated to only 9t

kareefardi, 2023-12-06:
@yunuseryilmaz18 @maliberty I agree that this is confusing and shouldn't happen. I will work on a fix.

PR #2062 (2023-12-26): Update `OpenROAD`
~ openroad_app -> `75f2f32`

## CI
~ Always compare regression and create reproducibles
~ Always escape design name
+ Add tests for attached crashes by adding issues reproducibles in designs submodules and symlink them to interactive scripts under ./tests
+ Fix failing designs in extended test:
  * salsa20: setup violations
  * y_huff: routing congestion
  * aes_core: pin antenna violations benchmark mismatch

---
[ci ets]
Fixes #2065
Fixes #2060 
Fixes #1935
Propagates https://github.com/The-OpenROAD-Project/OpenROAD/issues/3982
https://github.com/efabless/openlane-ci-design [... cut]
closed 2023-12-26 by donn