The-OpenROAD-Project/OpenLane #2063: [ERROR STA-0026] unterminated string constant

thesourcerer8, 2023-12-10:
### Description

There is a buffer overflow in read_liberty -corner Typical /home/philipp/libresilicon/StdCellLib/Catalog/gf180_stdcelllib_1/dependencies/pdks/gf180mcuD/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__tt_025C_5v00.lib
when reading liberty files with a function definition that is 67797 characters long. The string is properly terminated, but most likely due to the too small buffer, it thinks that the string constant isn't terminated.

### Expected Behavior

Well, 67790 characters is long, but it would be nice to have the scalability to even such very complex cells. I will also try to produce a more compact function description, but finding the right algorithms for that will take some time.

### Environment report

```shell
open_pdks e0f692f46654d6c7c99fc70a0c94a080dab53571
Kernel: Linux v6.2.0-39-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: 7ea7a2aeef4bea5445d133c26ca2b3e9f0d0c78f
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: e0f692f46654d6c7c99fc70a0c94a080dab53571, tested: dd7771c384ed36b91a25e9f8b314355fc26561be)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

7ea7a2a 2023-10-15T17:01:59+03:00 Fix `FP_PIN_ORDER_CFG` being overriden (#2017) - Kareem Farid -  (grafted, HEAD, tag: 2023.10.16)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/13629508/issue_reproducible.zip)

### Relevant log output

```shell
ERROR]: during executing openroad script /openlane/scripts/openroad/floorplan.tcl
[ERROR]: Log: ../home/philipp/libresilicon/StdCellLib/Catalog/gf180_stdcelllib_1/openlane/user_proj_example/runs/23_12_10_23_31/logs/floorplan/3-initial_fp.log
[ERROR]: Last 10 lines:
OpenROAD bdc8e944cd08ba8edebeca19fa531bd026690330 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
define_corners Typical
read_liberty -corner Typical /home/philipp/libresilicon/StdCellLib/Catalog/gf180_stdcelllib_1/dependencies/pdks/gf180mcuD/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__tt_025C_5v00.lib
read_liberty -corner Typical /home/philipp/libresilicon/StdCellLib/Catalog/gf180_stdcelllib_1/cells/lib/libresilicon.lib
[ERROR STA-0026] /home/philipp/libresilicon/StdCellLib/Catalog/gf180_stdcelllib_1/cells/lib/libresilicon.lib line 331, unterminated string constant.
Error: floorplan.tcl, 15 STA-0026
child process exited abnormally

[ERROR]: Creating issue reproducible...
```

maliberty, 2023-12-11:
What sort of cell has a "function definition that is 67797 characters long"?

maliberty, 2023-12-11:
This looks to be a flex limit on a single token.  However the expression is horrible unreduced and should be fixed.