The-OpenROAD-Project/OpenLane #2064: Command 'make test' not working after installing OpenLane

srichandrasrii, 2023-12-11:
### Description

Installed all the requirements like Docker, Python, git and make before installing OpenLane. Then I followed the following instructions present on the OpenLane github page.
```
cd $HOME
git clone https://github.com/The-OpenROAD-Project/OpenLane
cd OpenLane
make
make test
```
everything worked normal till 'make' command and i got following output
```
make[1]: Entering directory '/home/srichandra/OpenLane'
f691c8c0712ca6c6645e3fd548985b3cbcf08c78: Pulling from efabless/openlane
Digest: sha256:51727def16857065e353ea62176d50819d4d33b45950ca7411ef5bdf0f9bfa7a
Status: Image is up to date for efabless/openlane:f691c8c0712ca6c6645e3fd548985b3cbcf08c78
docker.io/efabless/openlane:f691c8c0712ca6c6645e3fd548985b3cbcf08c78

What's Next?
  1. Sign in to your Docker account → docker login
  2. View a summary of image vulnerabilities and recommendations → docker scout quickview efabless/openlane:f691c8c0712ca6c6645e3fd548985b3cbcf08c78
make[1]: Leaving directory '/home/srichandra/OpenLane'
PYTHONPATH= ./venv/bin/python3 -m pip install --upgrade --no-cache-dir 'volare>=0.12.3'
Requirement already satisfied: volare>=0.12.3 in ./venv/lib/python3.10/site-packages (0.15.2)
Requirement already satisfied: click<9,>=8.0.0 in ./venv/lib/python3.10/site-packages (from volare>=0.12.3) (8.1.7)
Requirement already satisfied: pyyaml<7,>=5 in ./venv/lib/python3.10/site-packages (from volare>=0.12.3) (6.0.1)
Requirement already satisfied: rich<14,>=12 in ./venv/lib/python3.10/site-packages (from volare>=0.12.3) (13.7.0)
Requirement already satisfied: httpx>=0.22.0 in ./venv/lib/python3.10/site-packages (from volare>=0.12.3) (0.25.2)
Requirement already satisfied: pcpp<2,>=1.2 in ./venv/lib/python3.10/site-packages (from volare>=0.12.3) (1.30)
Requirement already satisfied: zstandard<1,>=0.19.0 in ./venv/lib/python3.10/site-packages (from volare>=0.12.3) (0.22.0)
Requirement already satisfied: anyio in ./venv/lib/python3.10/site-packages (from httpx>=0.22.0->volare>=0.12.3) (4.1.0)
Requirement already satisfied: certifi in ./venv/lib/python3.10/site-packages (from httpx>=0.22.0->volare>=0.12.3) (2023.11.17)
Requirement already satisfied: httpcore==1.* in ./venv/lib/python3.10/site-packages (from httpx>=0.22.0->volare>=0.12.3) (1.0.2)
Requirement already satisfied: idna in ./venv/lib/python3.10/site-packages (from httpx>=0.22.0->volare>=0.12.3) (3.6)
Requirement already satisfied: sniffio in ./venv/lib/python3.10/site-packages (from httpx>=0.22.0->volare>=0.12.3) (1.3.0)
Requirement already satisfied: h11<0.15,>=0.13 in ./venv/lib/python3.10/site-packages (from httpcore==1.*->httpx>=0.22.0->volare>=0.12.3) (0.14.0)
Requirement already satisfied: markdown-it-py>=2.2.0 in ./venv/lib/python3.10/site-packages (from rich<14,>=12->volare>=0.12.3) (3.0.0)
Requirement already satisfied: pygments<3.0.0,>=2.13.0 in ./venv/lib/python3.10/site-packages (from rich<14,>=12->volare>=0.12.3) (2.17.2)
Requirement already satisfied: mdurl~=0.1 in ./venv/lib/python3.10/site-packages (from markdown-it-py>=2.2.0->rich<14,>=12->volare>=0.12.3) (0.1.2)
Requirement already satisfied: exceptiongroup>=1.0.2 in ./venv/lib/python3.10/site-packages (from anyio->httpx>=0.22.0->volare>=0.12.3) (1.2.0)
./venv/bin/volare enable --pdk sky130
Version e0f692f46654d6c7c99fc70a0c94a080dab53571 enabled for the sky130 
PDK.
```
But when I run the `make test` command it doesn't give the expected output.

[image]
I get this popup from docker and then the make command fails giving the following error.

```
cd /home/srichandra/OpenLane && \
	docker run --rm -v /home/srichandra:/home/srichandra -v /home/srichandra/OpenLane:/openlane -v /home/srichandra/OpenLane/empty:/openlane/install -v /home/srichandra/.volare:/home/srichandra/.volare -e PDK_ROOT=/home/srichandra/.volare -e PDK=sky130A  --user 1000:1000 -e DISPLAY=:1 -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/srichandra/.Xauthority:/.Xauthority --network host --security-opt seccomp=unconfined efabless/openlane:f691c8c0712ca6c6645e3fd548985b3cbcf08c78-amd64 sh -c "./flow.tcl -design spm -tag openlane_test -overwrite"
OpenLane f691c8c0712ca6c6645e3fd548985b3cbcf08c78
All rights reserved. (c) 2020-2022 Efabless Corporation and contributors.
Available under the Apache License, version 2.0. See the LICENSE file for more details.

can't create directory "/openlane/designs/spm/runs": permission denied
    while executing
"file mkdir $run_path"
    (procedure "prep" line 110)
    invoked from within
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 22)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 402)
make: *** [Makefile:204: test] Error 1
```

### Expected Behavior

As on the https://openlane.readthedocs.io/en/latest/getting_started/installation/installation_ubuntu.html page, this is the expected output when `make test` is run.
`Basic test passed`

### Environment report

```shell
srichandra@srichandra-Aspire-A715-75G:~/OpenLane$ python3 ./env.py issue-survey
open_pdks e0f692f46654d6c7c99fc70a0c94a080dab53571
Kernel: Linux v6.2.0-37-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.7 (OK)
OpenLane Git Version: f691c8c0712ca6c6645e3fd548985b3cbcf08c78
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f691c8c 2023-11-22T15:45:53+02:00 yosys/synth.tcl: migrate to newer supported drw & drf commands (#2051) - Darryl Miles -  (grafted, HEAD -> master, tag: 2023.11.23, origin/master, origin/HEAD)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

problem in installation and running the basic test run (spm)

### Relevant log output

```shell
already pasted in description section
```

donn, 2023-12-11:
I'll investigate- in the meantime, would you mind running `sestatus` for me in your terminal and reporting the result?

srichandrasrii, 2023-12-11:
> I'll investigate- in the meantime, would you mind running `sestatus` for me in your terminal and reporting the result?

Please find below the output when `sestatus` command is run:

```
srichandra@srichandra-Aspire-A715-75G:~/OpenLane$ sestatus
SELinux status:                 disabled
```

donn, 2023-12-11:
Thank you- that rules out SELinux.

jrossanderson, 2024-01-24:
I am having the same issue.  I ran `sestatus `as well, and the output is disabled.

donn, 2024-09-12:
We noticed you're using Docker Desktop- this adds another layer of permissions on top of the Docker engine. Please follow the documentation to install the Docker engine: https://openlane.readthedocs.io/en/latest/getting_started/installation/installation_ubuntu.html#docker-installation

closed 2024-09-12 by donn