The-OpenROAD-Project/OpenLane #2087: Using `SYNTH_PARAMETERS` in `config.tcl`

scorbetta, 2024-01-26:
Hello all, I'm having issues using the `SYNTH_PARAMETERS` option.

My Verilog design consists of a number of parametric modules. The top-level module has parameters as well, which get forwarded to the proper sub-modules. Some of the parameters are used in a `generate` statement.

The top-level module has 4 parameters with defaults:

```
NUM_INPUTS = 1
NUM_OUTPUTS = 1
WIDTH = 16
FRAC_BITS = 13
```

According to [this](https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/reference/configuration.md) the following `SYNTH_PARAMETERS` syntax is correct:

```
// From my config.tcl
set ::env(SYNTH_PARAMETERS) "NUM_INPUTS=1 NUM_OUTPUTS=1 WIDTH=8 FRAC_BITS=5"
```

However, the synthesis keeps complaining about ports having no drivers, then trims them and I eventually get an empty design, and the tool exits with an error. Please note that I have already checked all wires, ports, etc and everything is properly connected. So those warnings are wrong. Proof of this is that when I remove the above `SYNTH_PARAMETERS` option and change the defaults at the top-level to match those from the `SYNTH_PARAMETERS` option, everything goes as expected and synthesis finishes successfully.

It seems to me there are problems with the hierarchy when submodules too have parameters. I modified the synthesis script under `scripts/yosys` and appended the `hierarchy` command here and there, but with no luck.

Any idea?

Here's the output of `python3 ./env.py issue-survey`

```
Kernel: Linux v6.5.0-14-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: d054702b2cce04761cc2bc598f6b95c9d8ca7c6c
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

d054702 2023-07-19T16:09:15+03:00 remove `unset_propagated_clock` (#1908) - passant5 -  (grafted, HEAD, tag: 2023.07.19)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

Many thanks
S

vijayank88, 2024-01-27:
Pull latest commit and try again. 
Reproducible test case will narrow down to fix the issue quickly.

scorbetta, 2024-01-31:
It works with tag `2024.01.12`. Thanks

closed 2024-01-31 by scorbetta