The-OpenROAD-Project/OpenLane #2089: `FP_PIN_ORDER_CFG` not taken into account

scorbetta, 2024-01-31:
### Description

The contents from the file marked in `FP_PIN_ORDER_CFG` are not taken into accounts.

I am using the following:

```
// config.tcl
set script_dir $::env(DESIGN_DIR)
set ::end(FP_PIN_ORDER_CFG) "$script_dir/pin_order.cfg"
set ::env(FP_SIZING) "absolute"
set ::env(DIE_AREA) "0 0 500 2000"
```

```
// pin_order.cfg
#W
IN\[0\]
IN\[1\]
IN\[2\]
IN\[3\]

#E
OUT\[0\]
OUT\[1\]
OUT\[2\]
OUT\[3\]
OUT\[4\]
OUT\[5\]
OUT\[6\]
OUT\[7\]

#N

#S
```

The design has 4 inputs and 8 outputs. No matter what the contents of the `pin_order.cfg`, the placement is always random and sub-optimal.

### Expected Behavior

I would like to force few pins on some edges of the macro. In my case all `IN` bits on the West edge and all `OUT` bits on the East edge.

### Environment report

```shell
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v6.5.0-14-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: d054702b2cce04761cc2bc598f6b95c9d8ca7c6c
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

d054702 2023-07-19T16:09:15+03:00 remove `unset_propagated_clock` (#1908) - passant5 -  (grafted, HEAD, tag: 2023.07.19)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[WEIGHTS_XBAR.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/14109535/WEIGHTS_XBAR.tar.gz)

### Relevant log output

```shell
n/a
```

vijayank88, 2024-01-31:
Facing following error with your source and setup files.
```
[INFO]: Changing netlist to 'designs/WEIGHTS_XBAR/runs/run1/results/synthesis/WEIGHTS_XBAR.v'...
[ERROR]: The specified clock port 'CLK' does not exist in the top-level module.
[ERROR]: Step 1 (synthesis) failed with error:
-code 1 -level 0 -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_synthesis" line 33)
    invoked from within
"run_synthesis"} -errorline 1
```
Are you using `sky130` or `gf180` pdk?
you're using `2023.07.19` tag. Merge latest commit and try again.

scorbetta, 2024-01-31:
Hello, yes it's combinational. It's a XBAR-like macro I am trying to harden. I don't get that error, though. I am using `sky130`. Will try with latest commit.

scorbetta, 2024-01-31:
Nothing changed that much. Now pins are always on both `W` and `E` edges, but with random ordering. I'm using 2024.01.12 tag. I also added the `CLK` port.
[WEIGHTS_XBAR.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/14109927/WEIGHTS_XBAR.tar.gz)

vijayank88, 2024-01-31:
seems you typed wrong name
```
set ::end(FP_PIN_ORDER_CFG) "$script_dir/pin_order.cfg"
```
replace with
```
set ::env(FP_PIN_ORDER_CFG) "$script_dir/pin_order.cfg"
```
above change will resolve your issue

scorbetta, 2024-01-31:
Oh boy, silly me... I'm so sorry for that. Checked multiple times, never seen it...

Well, thanks for your time!

closed 2024-01-31 by scorbetta