The-OpenROAD-Project/OpenLane #2092: OpenLane deleting modules while synthesizing

Ashutosh-3107, 2024-02-16:
### Description

During flattening OpenLane deletes the modules and is unable to synthesisze and map it to standard library

### Expected Behavior

Not expecting to delete the modules while flattening

### Environment report

```shell
open_pdks cd1748bb197f9b7af62a54507de6624e30363943
Kernel: Linux v5.15.133.1-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: 9dbd8b5ea2bd891bed4dcc97df5c7439083f0368
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

9dbd8b5 2024-01-11T15:26:56+02:00 Fix `-synth_explore` crash (#2085) - Kareem Farid -  (grafted, HEAD -> master, tag: 2024.01.12, origin/master, origin/HEAD)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

https://github.com/merldsu/RISCV_Pipeline_Core 

### Relevant log output

```shell
/----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.34 (git sha1 4a1b5599258, gcc 8.3.1 -fPIC -Os)

[TCL: yosys -import] Command name collision: found pre-existing command `cd' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `eval' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `exec' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `read' -> skip.
[TCL: yosys -import] Command name collision: found pre-existing command `trace' -> skip.

1. Executing Verilog-2005 frontend: /openlane/designs/Pipeline_top/src/Hazard_unit.v
Parsing SystemVerilog input from `/openlane/designs/Pipeline_top/src/Hazard_unit.v' to AST representation.
Generating RTLIL representation for module `\hazard_unit'.
Successfully finished Verilog frontend.
[... 914 lines cut ...]
Checking module Pipeline_top...
Found and reported 0 problems.

78. Printing statistics.

=== Pipeline_top ===

   Number of wires:                  2
   Number of wire bits:              2
   Number of public wires:           2
   Number of public wire bits:       2
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  0

79. Executing Verilog backend.
Dumping module `\Pipeline_top'.

80. Executing JSON backend.

End of script. Logfile hash: 1f6996f824, CPU: user 0.37s system 0.00s, MEM: 28.44 MB peak
Yosys 0.34 (git sha1 4a1b5599258, gcc 8.3.1 -fPIC -Os)
Time spent: 28% 4x stat (0 sec), 25% 1x dfflibmap (0 sec), ...
```

kareefardi, 2024-02-18:
Can you post your configuration?

Ashutosh-3107, 2024-02-18:
> Can you post your configuration?

set ::env(DESIGN_NAME) {Pipeline_top}
set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/Hazard_unit.v $::env(DESIGN_DIR)/src/Data_Memory.v $::env(DESIGN_DIR)/src/ALU.v $::env(DESIGN_DIR)/src/Sign_Extend.v $::env(DESIGN_DIR)/src/Register_File.v $::env(DESIGN_DIR)/src/ALU_Decoder.v $::env(DESIGN_DIR)/src/Main_Decoder.v $::env(DESIGN_DIR)/src/Control_Unit_Top.v $::env(DESIGN_DIR)/src/Instruction_Memory.v $::env(DESIGN_DIR)/src/Mux.v $::env(DESIGN_DIR)/src/PC_Adder.v $::env(DESIGN_DIR)/src/PC.v $::env(DESIGN_DIR)/src/Writeback_Cycle.v $::env(DESIGN_DIR)/src/Memory_Cycle.v $::env(DESIGN_DIR)/src/Execute_Cycle.v $::env(DESIGN_DIR)/src/Decode_Cyle.v $::env(DESIGN_DIR)/src/Fetch_Cycle.v $::env(DESIGN_DIR)/src/Pipeline_top.v]
#set   ::env(SYNTH_HIERARCHICAL) 1
#set  ::env(SYNTH_FLAT_TOP) 0
#set ::env(SYNTH_STRATEGY) {DELAY 1}
#set ::env(SYNTH_NO_FLAT) 1
set ::env(SYNTH_EXPLORE) 0
set ::env(CLOCK_PORT) "clk"
set ::env(CLOCK_PERIOD) "80.0"

#set ::env(FP_PDN_MULTILAYER) {1}

set tech_specific_config "$::env(DESIGN_DIR)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl"
if { [file exists $tech_specific_config] == 1 } {
    source $tech_specific_config
}

kareefardi, 2024-02-22:
@Ashutosh-3107 The design has no output ports. There is nothing to drive by the top level module making everything unused and Yosys removes unused wires and cells.

closed 2024-02-29 by donn