The-OpenROAD-Project/OpenLane #2094: Synthesis warning for out of bounds signal

scorbetta, 2024-02-22:
### Description

I keep having the following warnings from the synthesis phase:

```
/openlane/designs/LAYER/src/HL_NEURON.v:95: Warning: Range select [15:8] out of bounds on signal `\weights': Setting all 8 result bits to undef.
//...
/openlane/designs/LAYER/src/HL_NEURON.v:109: Warning: Range select [127:120] out of bounds on signal `\weights': Setting all 8 result bits to undef.
```

This points to a signal named `weights` defined as follows in the referenced file:

```
wire [NUM_INPUTS*WIDTH-1:0] weights;
```

`NUM_INPUTS` and `WIDTH` parameters get values from the top-level. The top-level has the following default values:

```
parameter NUM_INPUTS = 16
parameter NUM_OUTPUTS = 8
parameter WIDTH = 8
parameter FRAC_BITS = 5
```

So that `weights` has actual dimension of 128 bits. Meaning the warning above is wrong. Strangely, the synthesis output `results/synthesis/HIDDEN_LAYER.v` contains the signal with the correct width.

The only way I found to bypass this is to add the following to the `config.tcl`:

```
set ::env(QUIT_ON_SYNTH_CHECKS) 0
```

But this is not the way to go. So what is the problem with the above case? Am I missing something?

Thanks
S

### Expected Behavior

Synthesis pass

### Environment report

```shell
open_pdks cd1748bb197f9b7af62a54507de6624e30363943
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v6.5.0-18-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 9dbd8b5ea2bd891bed4dcc97df5c7439083f0368
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

9dbd8b5e 2024-01-11T15:26:56+02:00 Fix `-synth_explore` crash (#2085) - Kareem Farid -  (HEAD, tag: 2024.01.12, origin/master, origin/HEAD, master)
a005df1f 2024-01-08T10:51:06+02:00 Updated link to newer version of spanish docs (#2082) - Laboratorio de Investigación en Microelectrónica y Arquitectura de Computadoras, EIE -- UCR -  (tag: 2024.01.09)
8bea8c51 2023-12-26T13:17:21+02:00 Update `OpenROAD` (#2062) - Kareem Farid -  (tag: 2023.12.27)
---
Git Remotes

origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

n/a

### Relevant log output

```shell
/openlane/designs/LAYER/src/HL_NEURON.v:95: Warning: Range select [15:8] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:96: Warning: Range select [23:16] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:97: Warning: Range select [31:24] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:98: Warning: Range select [39:32] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:99: Warning: Range select [47:40] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:100: Warning: Range select [55:48] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:101: Warning: Range select [63:56] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:102: Warning: Range select [71:64] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:103: Warning: Range select [79:72] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:104: Warning: Range select [87:80] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:105: Warning: Range select [95:88] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:106: Warning: Range select [103:96] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:107: Warning: Range select [111:104] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:108: Warning: Range select [119:112] out of bounds on signal `\weights': Setting all 8 result bits to undef.
/openlane/designs/LAYER/src/HL_NEURON.v:109: Warning: Range select [127:120] out of bounds on signal `\weights': Setting all 8 result bits to undef.
```
```

kareefardi, 2024-02-22:
@scorbetta Can you provide the RTL of your design?

scorbetta, 2024-02-22:
Sure thing. I'm attaching the OpenLANE config file as well. I had to remove the `runs` folder due to space.
[LAYER.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/14372574/LAYER.tar.gz)

scorbetta, 2024-02-29:
@kareefardi Any idea?

kareefardi, 2024-02-29:
@scorbetta Sorry I have been a bit busy this week. I am checking it out now.

scorbetta, 2024-02-29:
@kareefardi Got it. I can live with them then, however is this really how it should properly work? I mean, when I synthesize multiple Verilog files that belong to a given hierarchy, I would expect the top-level to forward values to the lower levels within the hierarchy. Otherwise there is no point to get a top with fixed parameters. And indeed that's how commercial tools do.

kareefardi, 2024-02-29:
@scorbetta I think we may be able to do that with `-defer` in Yosys when reading the Verilog files but this requires discussion internally on whether we want to do that. I will keep this issue open and update it.

scorbetta, 2024-03-01:
@kareefardi Indeed that's what I did. I am now using a patched `synth.tcl` script for synthesis. I added the `-defer` option to the `read_verilog` commands, and placed a `hierarchy -check -top $vtop` command later. Warnings go away.

PR #2100 (2024-03-11): Defer compilation of AST of verilog files read in by Yosys till all the files are read
Add `-defer` to read_verilog and run `hierarchy -check -top $vtop` after all the Verilog files are read 

---

Fixes #2094
closed 2024-03-11 by donn