The-OpenROAD-Project/OpenLane #2095: unable to view the floorplan def files in klayout of oprn lane

Ashutosh-3107, 2024-02-26:
### Description

ERROR: Unable to open file: /home/ashutosh_anand/OpenLane/designs/Pipeline_top/runs/run1/results/floorplan/Pipeline_top.def/merged.lef (errno=20)

### Expected Behavior

to be able to view the intermediate lef / def files

### Environment report

```shell
open_pdks cd1748bb197f9b7af62a54507de6624e30363943
Kernel: Linux v5.15.133.1-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: 9dbd8b5ea2bd891bed4dcc97df5c7439083f0368
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

9dbd8b5 2024-01-11T15:26:56+02:00 Fix `-synth_explore` crash (#2085) - Kareem Farid -  (grafted, HEAD -> master, tag: 2024.01.12, origin/master, origin/HEAD)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[Pipeline_top.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/14409291/Pipeline_top.zip)

### Relevant log output

```shell
ERROR: Unable to open file: /home/ashutosh_anand/OpenLane/designs/Pipeline_top/runs/run1/results/floorplan/Pipeline_top.def/merged.lef (errno=20)
```

d-m-bailey, 2024-02-27:
@Ashutosh-3107 You can not directly open a DEF file by itself. https://www.klayout.de/doc/about/lef_def_import.html
Try using the File -> Import -> DEF/LEF dialog.

d-m-bailey, 2024-02-27:
Looks like you can also specify the LEF file in Tools -> Manage Technologies -> Reader Options -> LEF/DEF.

Ashutosh-3107, 2024-02-27:
> Looks like you can also specify the LEF file in Tools -> Manage Technologies -> Reader Options -> LEF/DEF.

tried both the methods still the same error pops

d-m-bailey, 2024-02-27:
@Ashutosh-3107 What LEF file are you using?

Ashutosh-3107, 2024-02-27:
> @Ashutosh-3107 What LEF file are you using?

I have uploaded in the Reproduction material

d-m-bailey, 2024-02-27:
`Pipeline_top.zip` only contains the DEF data. What LEF file are you using?

Ashutosh-3107, 2024-02-27:
I am unsure which lef file to use

> only contains the DEF data. What LEF file are you using?

d-m-bailey, 2024-02-27:
Can you send a screen shot of your reader options?

The DEF file contains routing information and cell references. The data for the actual cells is in the LEF file. There are different types of LEF files. A technology LEF file will only contain wire sizes, spacing, etc. - it probably doesn't have the LEF data for cells that you'll need to view in klayout.

Ashutosh-3107, 2024-02-27:
[image]
[image]
[image]

d-m-bailey, 2024-02-27:
I doubt you need all 3 files, but I don't know if it causes a problem.

Can you `grep sky130_fd_sc_hd__dfrtp_2 <lef_file>`?

Ashutosh-3107, 2024-02-27:
> I doubt you need all 3 files, but I don't know if it causes a problem.
> 
> Can you `grep sky130_fd_sc_hd__dfrtp_2 <lef_file>`?

[image]
[image]
[image]
I have included all the related files still it is popping the same error

d-m-bailey, 2024-02-28:
@Ashutosh-3107 I'm using a different pdk version, but I'm able to open the def file with only specifying the one lef file from the pdk. Can you try with just the `sky130_fd_sc_hd.lef` file from the pdk?

I wonder if the klayout version makes a difference. I'm using 0.28.15.

Are you using docker for either klayout or file references?

Ashutosh-3107, 2024-02-28:
> @Ashutosh-3107 I'm using a different pdk version, but I'm able to open the def file with only specifying the one lef file from the pdk. Can you try with just the `sky130_fd_sc_hd.lef` file from the [...]

I am using the docker, may be there is bug in the klayout app, do you have any idea how to fix this klayout. I am currently using 0.28.2

d-m-bailey, 2024-02-28:
You might try installing klayout locally and see if you can open the def file there. I don’t know how to update the docker version.

vijayank88, 2024-02-28:
@Ashutosh-3107 

I tried for spm design with floorplan `spm.def` with KLayout loading properly.
[image]

You can try adding in LEF files space only with `merged.nom.lef` and remove other lef files and try to load it.

Ashutosh-3107, 2024-02-28:
> @Ashutosh-3107
> 
> I tried for spm design with floorplan `spm.def` with KLayout loading properly. ![Load_def_with_KLayout](https://private-user-images.githubusercontent.com/85787913/308521822-c5ca500a-16de-4fdc-9412-09d9af2eb221.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxMjIyNzAsIm5iZiI6MTcwOTEyMTk3MCwicGF0aCI6Ii84NTc4NzkxMy8zMDg1MjE4MjItYzVjYTUwMGEtMTZkZS00ZmRjLTk0MTItMDlkOWFmMmViMjIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI4VDEyMDYxMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlmOTFiOWYzODRlZDA0MTI1MWUzYzg4YjgwYTk5MGFiZWQ3ZTQyYjdkYzEzNGEwMTQwY2ZmYjkxYzhhMjg4YjQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4sD-keWC11xvfNLAjJi8-Y_Q4duh3U_v4VL4qiztyiI)
> 
> You can try adding in LEF files space only with `merged.nom.lef` and remove other lef files and try to load it.

[image]
[image]

I guess there is a bug in the klayout of the openlane. If there is any method to remove the bug or update klayout please recommend.

stefanottili, 2024-03-02:
I'm using klayout with lef+def without any issues, but what's the point ?
Openroad is much better at viewing/debugging lef/def files then klayout.
You have netlist and timing information and placement/routing/power maps.
```
openroad -gui load_design.or
```
with load_design.or containing
```
read_lef tech.lef
read_lef macro.lef
read_def design.lef
```

donn, 2024-09-12:
Staled out.

closed 2024-09-12 by donn