The-OpenROAD-Project/OpenLane #2096: [ERROR PDN-0179] Unable to repair all channels.

scorbetta, 2024-02-28:
### Description

PDN phase generates the following error:

```
[ERROR PDN-0179] Unable to repair all channels.
```

after generating a bunch of warnings:

```
[WARNING PDN-0178] Remaining channel (635.2600, 146.6400) - (639.8600, 158.0000) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (635.2600, 282.6400) - (639.8600, 294.0000) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (635.2600, 445.8400) - (639.8600, 457.2000) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (635.2600, 581.8400) - (639.8600, 593.2000) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (635.2600, 978.9600) - (639.8600, 990.3200) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (635.2600, 1114.9600) - (639.8600, 1126.3200) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (935.1800, 1150.3200) - (939.7800, 1161.6800) on met1 for nets: vccd1
```

The manual macro placement seems ok, since there's no overlapping. The flow (up to the PDN phase) works if I leave the macro placement automatic (with the addition of `PL_BASIC_PLACEMENT`).

### Expected Behavior

.

### Environment report

```shell
open_pdks cd1748bb197f9b7af62a54507de6624e30363943
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v6.5.0-18-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 9dbd8b5ea2bd891bed4dcc97df5c7439083f0368
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

9dbd8b5e 2024-01-11T15:26:56+02:00 Fix `-synth_explore` crash (#2085) - Kareem Farid -  (HEAD, tag: 2024.01.12, origin/master, origin/HEAD, master)
a005df1f 2024-01-08T10:51:06+02:00 Updated link to newer version of spanish docs (#2082) - Laboratorio de Investigación en Microelectrónica y Arquitectura de Computadoras, EIE -- UCR -  (tag: 2024.01.09)
8bea8c51 2023-12-26T13:17:21+02:00 Update `OpenROAD` (#2062) - Kareem Farid -  (tag: 2023.12.27)
---
Git Remotes

origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/14437027/issue_reproducible.tar.gz)

### Relevant log output

```shell
Contents of `runs/run1/logs/floorplan/7-pdn.log`

OpenROAD 75f2f325b7a42e56a92404f33af8e96530d9b202 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Reading ODB at '/openlane/designs/HIDDEN_LAYER/runs/run1/tmp/floorplan/6-tapcell.odb'…
define_corners Typical
read_liberty -corner Typical /home/scorbetta/work/tools/pdk/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
read_liberty -corner Typical /openlane/designs/HIDDEN_LAYER/../HL_NEURON/runs/run1//results/final/lib/HL_NEURON.lib
Using 1e-12 for capacitance...
Using 1e+03 for resistance...
Using 1e-09 for time...
Using 1e+00 for voltage...
Using 1e-03 for current...
Using 1e-09 for power...
Using 1e-06 for distance...
Reading design constraints file at '/openlane/scripts/base.sdc'…
[INFO]: Setting output delay to: 4.0
[INFO]: Setting input delay to: 4.0
[INFO]: Setting load to: 0.033442
[INFO]: Setting clock uncertainty to: 0.25
[INFO]: Setting clock transition to: 0.15
[INFO]: Setting timing derate to: 5.0 %
[INFO PDN-0001] Inserting grid: stdcell_grid
[WARNING PDN-0178] Remaining channel (335.3400, 146.6400) - (339.9400, 158.0000) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (335.3400, 282.6400) - (339.9400, 294.0000) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (335.3400, 445.8400) - (339.9400, 457.2000) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (335.3400, 581.8400) - (339.9400, 593.2000) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (335.3400, 978.9600) - (339.9400, 990.3200) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (335.3400, 1114.9600) - (339.9400, 1126.3200) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (635.2600, 146.6400) - (639.8600, 158.0000) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (635.2600, 282.6400) - (639.8600, 294.0000) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (635.2600, 445.8400) - (639.8600, 457.2000) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (635.2600, 581.8400) - (639.8600, 593.2000) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (635.2600, 978.9600) - (639.8600, 990.3200) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (635.2600, 1114.9600) - (639.8600, 1126.3200) on met1 for nets: vssd1
[WARNING PDN-0178] Remaining channel (935.1800, 1150.3200) - (939.7800, 1161.6800) on met1 for nets: vccd1
[ERROR PDN-0179] Unable to repair all channels.
PDN-0179
```
```

d-m-bailey, 2024-02-28:
@scorbetta looks like you're trying to place hard macros. Does your config file have `EXTRA_GDS_FILES` and `EXTRA_LEFs` defined for the `HL_NEURON` macro?

scorbetta, 2024-02-29:
@d-m-bailey Yes. As I said, the flow works without the manual macro placement. And yet manual macro placement seems correct since there is no overlapping.

kareefardi, 2024-02-29:
@scorbetta This is one of the reported locations by OpenROAD. [image]
In the decap cells between the macros, if there isn't a vertical stripe passing through they wouldn't be powered and cause such an error. You may want to increase FP_TAP_HORIZONTAL_HALO such that these cells wouldn't exist in these  location.

scorbetta, 2024-02-29:
@kareefardi Perfect! Many thanks for your help. It indeed is consistent with the way the tool places the macros automatically, so close each other that there is no space to be filled with decaps.

closed 2024-02-29 by scorbetta