The-OpenROAD-Project/OpenLane #2098: Wire is used but has no driver

scorbetta, 2024-03-01:
### Description

I am running into the following warning at synthesis time:

```
Warning: Wire HL_NEURON.\VALUE_OUT [7] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [6] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [5] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [4] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [3] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [2] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [1] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [0] is used but has no driver.
Warning: Wire HL_NEURON.\VALID_OUT is used but has no driver.
Warning: Wire HL_NEURON.\READY is used but has no driver.
Warning: Wire HL_NEURON.\OVERFLOW is used but has no driver.
```

but I don't understand where this is wrong. Simulation shows that wires are properly connected. Also, the synthesized netlist `results/synthesis/*.v` seems correct, since the above signals are properly driven.

### Expected Behavior

No warning reported by the tool.

### Environment report

```shell
open_pdks cd1748bb197f9b7af62a54507de6624e30363943
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v6.5.0-21-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 9dbd8b5ea2bd891bed4dcc97df5c7439083f0368
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

9dbd8b5e 2024-01-11T15:26:56+02:00 Fix `-synth_explore` crash (#2085) - Kareem Farid -  (HEAD, tag: 2024.01.12, origin/master, origin/HEAD, master)
a005df1f 2024-01-08T10:51:06+02:00 Updated link to newer version of spanish docs (#2082) - Laboratorio de Investigación en Microelectrónica y Arquitectura de Computadoras, EIE -- UCR -  (tag: 2024.01.09)
8bea8c51 2023-12-26T13:17:21+02:00 Update `OpenROAD` (#2062) - Kareem Farid -  (tag: 2023.12.27)
---
Git Remotes

origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

Contents of the `design/HL_NEURON` folder for OpenLane

[HL_NEURON.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/14459132/HL_NEURON.tar.gz)

### Relevant log output

```shell
Warning: Wire HL_NEURON.\VALUE_OUT [7] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [6] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [5] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [4] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [3] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [2] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [1] is used but has no driver.
Warning: Wire HL_NEURON.\VALUE_OUT [0] is used but has no driver.
Warning: Wire HL_NEURON.\VALID_OUT is used but has no driver.
Warning: Wire HL_NEURON.\READY is used but has no driver.
Warning: Wire HL_NEURON.\OVERFLOW is used but has no driver.
```

kareefardi, 2024-03-10:
@scorbetta This is due to OpenLane 1 not reading in the PDK liberty files for Yosys itself (they are being read in other parts for synthesis such as abc and during calculating the design area). At the final check pass, the std cells are (sort of?) not defined and you would find all output ports, driven by std cells, spitting out this warning. This was addressed in OpenLane 2. Thanks for the issue report, we will address it in OpenLane 1 as well.

scorbetta, 2024-03-11:
@kareefardi Thanks for the answer. Is this still a problem for the phases following synthesis? I mean, can I trust the output of the entire Openlane 1 flow at the end? Thanks

kareefardi, 2024-03-12:
> @kareefardi Thanks for the answer. Is this still a problem for the phases following synthesis? I mean, can I trust the output of the entire Openlane 1 flow at the end? Thanks

No these kind of warnings for the final checks in Yosys are harmless and should be removed by the attached PR.

closed 2024-03-14 by donn