The-OpenROAD-Project/OpenLane #2101: issues in using the floorplan core utilization command

Ashutosh-3107, 2024-03-11:
### Description

Even though the default value  of floorplan core utilization is 50 percent of the die area , while running the floorplan core utilization to 0.6 the core almost utilizes the entire die area which leaves very less place for routing stage

### Expected Behavior

to utilize the area as set in the configuration tcl file

### Environment report

```shell
open_pdks cd1748bb197f9b7af62a54507de6624e30363943
Kernel: Linux v5.15.146.1-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: 9dbd8b5ea2bd891bed4dcc97df5c7439083f0368
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

9dbd8b5 2024-01-11T15:26:56+02:00 Fix `-synth_explore` crash (#2085) - Kareem Farid -  (grafted, HEAD -> master, tag: 2024.01.12, origin/master, origin/HEAD)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[reproduction_material.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/14553552/reproduction_material.zip)

src rtl along with configuration tcl

### Relevant log output

```shell
floorplan core utilization according to the configuration tcl
```

vijayank88, 2024-03-11:
`set ::env(FP_CORE_UTIL) 0.7` update to `set ::env(FP_CORE_UTIL) 40` and try.
Core utilization in the range of 30-70 value normally and remaining space for routing.

Ashutosh-3107, 2024-03-11:
> `set ::env(FP_CORE_UTIL) 0.7` update to `set ::env(FP_CORE_UTIL) 40` and try.
> Core utilization in the range of 30-70 value normally and remaining space for routing.

[INFO] Floorplanned on a die area of 0.0 0.0 1986.82 1997.54 (microns). Saving to /openlane/designs/Pipeline_top/runs/run1/reports/floorplan/3-initial_fp_die_area.rpt.
[INFO] Floorplanned on a core area of 5.52 10.88 1981.22 1985.6 (microns). Saving to /openlane/designs/Pipeline_top/runs/run1/reports/floorplan/3-initial_fp_core_area.rpt.

It still utilizes most of the of the die area . I am doubting that if there is a gap in my understanding of how the command works

vijayank88, 2024-03-11:
Continue the flow and it will throw an error if more congestion. Check where it is stops now DPL/GRT

kareefardi, 2024-03-11:
@Ashutosh-3107 Setting `FP_CORE_UTIL` affects both die area and core area. Core utilization is not the calculated relative to the die area. It is the area utilized by cells and macros of the deisgn inside the core (area). Die area is the core area with additional margins that are configured using `BOTTOM_MARGIN_MULT`, `TOP_MARGIN_MULT`, `LEFT_MARGIN_MULT`, and `RIGHT_MARGIN_MULT`

donn, 2024-09-12:
Staled out.

closed 2024-09-12 by donn