The-OpenROAD-Project/OpenLane #2102: No via inserted -> Unconnected PDN node

scorbetta, 2024-03-11:
### Description

My design consists of two hierarchical blocks `MACRO` and `TOP`. The `MACRO` block is hardened first, with power ring enabled; the flow is then run on `TOP` block which instantiates one instance of `MACRO`.

The following messages appear:

```
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Log: designs/TEST_MACRO_TOP/runs/run1/logs/floorplan/7-pdn.log
[ERROR]: Last 10 lines:
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 293.440um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 230.710um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 230.710um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 77.530um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 77.530um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 54.080um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 54.080um), layer: met5.
[ERROR PSM-0069] Check connectivity failed.
Error: pdn.tcl, 43 PSM-0069
child process exited abnormally

[ERROR]: Creating issue reproducible...
```

### Expected Behavior

Power ring properly connected.

### Environment report

```shell
open_pdks cd1748bb197f9b7af62a54507de6624e30363943
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v6.5.0-25-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 9dbd8b5ea2bd891bed4dcc97df5c7439083f0368
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

9dbd8b5e 2024-01-11T15:26:56+02:00 Fix `-synth_explore` crash (#2085) - Kareem Farid -  (HEAD, tag: 2024.01.12, origin/master, origin/HEAD, master)
a005df1f 2024-01-08T10:51:06+02:00 Updated link to newer version of spanish docs (#2082) - Laboratorio de Investigación en Microelectrónica y Arquitectura de Computadoras, EIE -- UCR -  (tag: 2024.01.09)
8bea8c51 2023-12-26T13:17:21+02:00 Update `OpenROAD` (#2062) - Kareem Farid -  (tag: 2023.12.27)
---
Git Remotes

origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/14556215/issue_reproducible.tar.gz)

### Relevant log output

```shell
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (301.060um, 293.440um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (301.060um, 293.440um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (301.060um, 230.710um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (301.060um, 230.710um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (301.060um, 77.530um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (301.060um, 77.530um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (301.060um, 54.080um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (301.060um, 54.080um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (225.440um, 293.440um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (225.440um, 293.440um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (225.440um, 230.710um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (225.440um, 230.710um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (225.440um, 77.530um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (225.440um, 77.530um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (225.440um, 54.080um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (225.440um, 54.080um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (71.840um, 293.440um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (71.840um, 293.440um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (71.840um, 230.710um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (71.840um, 230.710um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (71.840um, 77.530um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (71.840um, 77.530um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (71.840um, 54.080um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (71.840um, 54.080um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 293.440um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 293.440um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 230.710um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 230.710um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 77.530um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 77.530um), layer: met5.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 54.080um), layer: met4.
[WARNING PSM-0038] Unconnected PDN node on net vccd1 at location (48.720um, 54.080um), layer: met5.
[ERROR PSM-0069] Check connectivity failed.
```

kareefardi, 2024-03-17:
@scorbetta I think this is a false positive by `check_power_grid`. There are vias at the corners of the power ring of the macro, where the tool is complaining, and it seems to be connected. Can you continue the flow with `FP_PDN_CHECK_NODES` disabled and see if you get any other errors?

kareefardi, 2024-03-18:
@scorbetta Great. I will keep this issue open to better understand the false positive from OpenROAD.

closed 2024-07-04 by kareefardi