The-OpenROAD-Project/OpenLane #2104: ldpcEncDec Step 18 18-resizer_design.log failed

manish03, 2024-03-21:
### Description

There is routing congestion while running step 18

### Expected Behavior

Need to complete step 18 with environment variables
        "FP_CORE_UTIL": 20,
        "PL_TARGET_DENSITY": 0.45,

### Environment report

```shell
parallels@ubuntu-linux-22-04-02-desktop:...hcb/asic/openlane% python3 ./env.py issue-survey
Kernel: Linux v5.15.0-101-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: 30ee1388932eb55a89ad84ee43997bfe3a386421
NOT FOUND: Please install pip using your operating system's package manager.
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

30ee138 2023-11-10T05:03:43+00:00 Update pyyaml - Mohamed Gaber -  (grafted, HEAD, tag: 2023.07.19-1)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

make ldpcEncDec
[issue_ldpcEncDec.18a.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/14686677/issue_ldpcEncDec.18a.tar.gz)
[issue_ldpcEncDec.18b.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/14686688/issue_ldpcEncDec.18b.tar.gz)
[issue_ldpcEncDec.18c.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/14686698/issue_ldpcEncDec.18c.tar.gz)
[issue_ldpcEncDec.18d.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/14686704/issue_ldpcEncDec.18d.tar.gz)

command used
split -b 24M issue_ldpcEncDec.18.tar.gz issue_ldpcEncDec.18.tar.gz.part
mv issue_ldpcEncDec.18.tar.gz.partaa issue_ldpcEncDec.18a.tar.gz
mv issue_ldpcEncDec.18.tar.gz.partab issue_ldpcEncDec.18b.tar.gz
mv issue_ldpcEncDec.18.tar.gz.partac issue_ldpcEncDec.18c.tar.gz
mv issue_ldpcEncDec.18.tar.gz.partad issue_ldpcEncDec.18d.tar.gz

### Relevant log output

```shell
STEP 18]
[INFO]: Running Global Routing Resizer Design Optimizations (log: ../media/psf/Home/Downloads/sljt1.comp.xfer/backupcvs/hcb1/hcb/LDPCENCDEC/openlane/ldpcEncDec/runs/24_03_20_20_07/logs/routing/18-resizer_design.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/resizer_routing_design.tcl
[ERROR]: Log: ../media/psf/Home/Downloads/sljt1.comp.xfer/backupcvs/hcb1/hcb/LDPCENCDEC/openlane/ldpcEncDec/runs/24_03_20_20_07/logs/routing/18-resizer_design.log
[ERROR]: Last 10 lines:
[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[INFO GRT-0197] Via related to pin nodes: 706697
[INFO GRT-0198] Via related Steiner nodes: 23502
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 830635
[INFO GRT-0112] Final usage 3D: 3883704
[ERROR GRT-0119] Routing congestion too high. Check the congestion heatmap in the GUI and load /media/psf/Home/Downloads/sljt1.comp.xfer/backupcvs/hcb1/hcb/LDPCENCDEC/openlane/ldpcEncDec/runs/24_03_20_20_07/tmp/routing/resizer-routing-design-congestion.rpt in the DRC viewer.
Error: resizer_routing_design.tcl, 46 GRT-0119
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.
```

vijayank88, 2024-03-25:
While running latest OpenROAD facing crash during `gui::show`.
```
[INFO GRT-0112] Final usage 3D: 5022765
[ERROR GRT-0119] Routing congestion too high. Check the congestion heatmap in the GUI and load ./tmp/routing/resizer-routing-design-congestion.rpt in the DRC viewer.
Error: resizer_routing_design.tcl, 46 GRT-0119
while evaluating {source run.tcl}
openroad> gui::show
[WARNING GUI-0076] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-'
process 10: The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.
Most likely, the application was supposed to call dbus_connection_close(), since this is a private connection.
  D-Bus not built with -rdynamic so unable to print a backtrace
Signal 6 received
Stack trace:
 0# 0x0000000000D81E07 in openroad
 1# 0x00007F5BDCC82400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x00007F5BCE2D3655 in /lib64/libdbus-1.so.3
 5# _dbus_warn_check_failed in /lib64/libdbus-1.so.3
 6# 0x00007F5BD395FBFB in /lib64/libQt5DBus.so.5
 7# QObject::event(QEvent*) in /lib64/libQt5Core.so.5
 8# 0x00007F5BDDFB2A55 in /lib64/libQt5Core.so.5
 9# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib64/libQt5Core.so.5
10# QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) in /lib64/libQt5Core.so.5
11# 0x00007F5BDE001A03 in /lib64/libQt5Core.so.5
12# g_main_context_dispatch in /lib64/libglib-2.0.so.0
13# 0x00007F5BD99E83F8 in /lib64/libglib-2.0.so.0
14# g_main_context_iteration in /lib64/libglib-2.0.so.0
15# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
16# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
17# QThread::exec() in /lib64/libQt5Core.so.5
18# 0x00007F5BD395D3B5 in /lib64/libQt5DBus.so.5
19# 0x00007F5BDDE08E71 in /lib64/libQt5Core.so.5
20# 0x00007F5BE156DEA5 in /lib64/libpthread.so.0
21# clone in /lib64/libc.so.6
Aborted (core dumped)
```
@manish03 can you attach screenshot of placement stage layout view?

manish03, 2024-03-29:
Hi,
I have attached an image of the placement def. The run logs and defs are
also uploaded on google drive. This link
<https://drive.google.com/file/d/1Rai2BjUs7r5tKEHdz4EyO71axhOrBqj1/view?usp=sharing>
is shared on invite only.
This run is different from the one you have.
Let me know if the die size is not able to have this design, causing
congestion.
Thanks,
Manish

manish03, 2024-03-29:
HiPlease use tar xjf filename.tjzTo untarThanksManishSent from my iPhone

<ldpcEncDec.png>

vijayank88, 2024-03-29:
attach the file in .zip or .tgz format. @manish03

manish03, 2024-03-29:
Hi,

 I have shared the link
<https://drive.google.com/file/d/1hzp49vh1jpvEdIA5iR5K9x2LyjfpDRhY/view?usp=drive_link>
with you.

tar xzvf 24_03_24_23_40.tgz

Thanks & Regards
Manish

vijayank88, 2024-03-29:
@manish03 
I just checked your config.json. It is missing `"FP_PDN_MACRO_HOOKS"` definition.
Refer https://openlane.readthedocs.io/en/latest/tutorials/openram.html this for configuration and how to integrate SRAM macros.
Is it going for fab?

manish03, 2024-03-30:
Hi,
 I still have the routing Congestion. I am not sure of the utilization.
From Synthesis report I get area as 2901153.668000
In the picture there is very little space left
Can tiny tapeout be an option as. I can add more die as needed

link
<https://drive.google.com/file/d/1z6SGmjC7LzVnmjZ2L9TPr-_BLM71FYtl/view?usp=sharing>

kareefardi, 2024-06-25:
@vijayank88 There is still a bug with OpenROAD for the GUI crashing. I will file a separate issue with OpenRAOD 

@manish03 I viewed the layout using KLayout and indeed the design appears congested. There are macros at every corner and the standard cells are concentrated in the middle. Pins are also distributed at the corners which might prevent. You might want to change the pin placement and manually place the macros in better places such that the macro pins and design pins are easily accessible and don't cause congestion. You might also want to spread the standard cells a bit.

donn, 2024-09-12:
Staled out.

closed 2024-09-12 by donn