The-OpenROAD-Project/OpenLane #2106: Multiple conflicting drivers

Aadhithya33, 2024-03-30:
### Description

When i synthesized this [verilog code](https://github.com/vinayrayapati/rv32i/blob/main/iiitb_rv32i.v) in yosys, it was successful.But when i ran it in openlane i got an warning (Multiple conflicting driver). Even i got the same warning when i synthesized using yosys but it was resolved by yosys. But in openlane it did not.
[image]
[image]

### Expected Behavior

Ideally multiple conflicting driver should be resolved but i am getting an warning/error .

### Environment report

```shell
open_pdks cd1748bb197f9b7af62a54507de6624e30363943
Kernel: Linux v6.5.0-21-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v26.0.0 (OK)
OpenLane Git Version: 2719508edbb579ab7317d1e0942b774d8d4aa214
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

2719508e 2024-03-14T14:41:49+02:00 Remove SYNTH_READ_BLACKBOX_LIB (#2099) - Kareem Farid -  (HEAD -> master, tag: 2024.03.15, origin/master, origin/HEAD)
525d7ed3 2024-03-11T13:31:44+02:00 Defer AST Compilation in Yosys (#2100) - Kareem Farid -  (tag: 2024.03.12)
a663df28 2024-03-06T15:47:26+02:00 Update OpenROAD (#2093) - Mohamed Gaber -  (tag: 2024.03.07)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[iiitb_rv32i.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/14813122/iiitb_rv32i.zip)

### Relevant log output

```shell
19. Executing CHECK pass (checking for obvious problems).
Checking module iiitb_rv32i...
Warning: multiple conflicting drivers for iiitb_rv32i.\BR_EN:
    port Q[0] of cell $procdff$2403 ($adff)
    port Q[0] of cell $procdff$2377 ($dff)
Warning: multiple conflicting drivers for iiitb_rv32i.\REG[0] [31]:
    port Q[31] of cell $procdff$2411 ($adff)
    port Q[31] of cell $procdff$2331 ($dff)
Warning: multiple conflicting drivers for iiitb_rv32i.\REG[0] [30]:
    port Q[30] of cell $procdff$2411 ($adff)
    port Q[30] of cell $procdff$2331 ($dff)
Warning: multiple conflicting drivers for iiitb_rv32i.\REG[0] [29]:
    port Q[29] of cell $procdff$2411 ($adff)
    port Q[29] of cell $procdff$2331 ($dff)
Warning: multiple conflicting drivers for iiitb_rv32i.\REG[0] [28]:
    port Q[28] of cell $procdff$2411 ($adff)
    port Q[28] of cell $procdff$2331 ($dff)
Warning: multiple conflicting drivers for iiitb_rv32i.\REG[0] [27]:
    port Q[27] of cell $procdff$2411 ($adff)
    port Q[27] of cell $procdff$2331 ($dff)
```

kareefardi, 2024-03-31:
@Aadhithya33 The warning from Yosys does get handled in OpenLane. If you look in the log file, you will find the following:
```
Warning: Driver-driver conflict for \BR_EN between cell $procdff$2377.Q and constant 1'0 in iiitb_rv32i: Resolved using constant.
Warning: Driver-driver conflict for \REG[6] [31] between cell $procdff$2337.Q and constant 1'0 in iiitb_rv32i: Resolved using constant.
Warning: Driver-driver conflict for \REG[6] [30] between cell $procdff$2337.Q and constant 1'0 in iiitb_rv32i: Resolved using constant.
```

However, OpenLane consider this behavior dangerous, captures the original multiple conflicting drivers warning from Yosys and throws an error. This behavior is controlled by the variable `QUIT_ON_SYNTH_CHECKS`

Aadhithya33, 2024-03-31:
Thank you for the reply. 
Which log file are you talking about ? I did not find anything like this in openlane but i found these warnings getting resolved  when i ran my design in yosys.
Is it okay to do this "set ::env(QUIT_ON_SYNTH_CHECKS) 0" .

Aadhithya33, 2024-03-31:
@kareefardi Thank you for the reply.
Which log file are you talking about ? I did not find anything like this in openlane but i found these warnings getting resolved when i ran my design in yosys.
Is it okay to do this "set ::env(QUIT_ON_SYNTH_CHECKS) 0"

Aadhithya33, 2024-03-31:
@kareefardi Thank you for the reply.
Is it okay to do this "set ::env(QUIT_ON_SYNTH_CHECKS) 0" ?
Can i ignore these warnings and proceed to PnR stage?

kareefardi, 2024-04-02:
@Aadhithya33 I think you should try to fix the errors/warnings in your RTL first.

Aadhithya33, 2024-04-02:
@kareefardi okay thank you

closed 2024-04-03 by kareefardi