The-OpenROAD-Project/OpenLane #2107: Add python `libprase` dependency to `requirements.txt`

dlmiles, 2024-03-31:
### Description

$ find . -type f -name "*.py" -exec fgrep -Hn libparse {} \; | grep import
./scripts/utils/get_default_operating_conditions.py:18:from libparse import LibertyParser

Please add this library to the appropriate requirements.txt file, I have seen it needed at runtime when using the flow to generate GDS.  

0.3.1 appears to be the version I need to manually install so I do not see a late stage flow failure.

```
libparse>=0.3.1
```

### Expected Behavior

I do not expect the flow runtime to fail in the later stages with missing Python module errors concerning libparse because it was not installed.

### Environment report

```shell
All OL tags for the past 6+ months appear to be affected.
```

### Reproduction material

All OL tags for the past 6+ months appear to be affected.

### Relevant log output

```shell
Not to hand, but an audit of the `*.py` and the `import` declarations should be enough to see the dependency exists.
```

donn, 2024-03-31:
Thanks for catching that. I'm on it.

donn, 2024-03-31:
So-- this is something of an idiosyncrasy of OpenLane 1's but long story short, `requirements.txt` are primarily concerned with people installing OpenLane, not using it. The runtime requirements are in `dependencies/python/run_time.txt`.

I'll go ahead and update it while I'm at it. Sorry for the confusion.

dlmiles, 2024-03-31:
Sure I did see there were 5 `requirements.txt` like files in the tree.  I use the term in the generic Python sense of the mechanism.

Hence my use of the term `Please add this library to the appropriate requirements.txt file` as I do not know the individual details of all 5.

dlmiles, 2024-04-03:
This is how things look to me without the change.

```
$ pip list
WARNING: The directory '/openlane/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you should use sudo's -H flag.
Package            Version
------------------ -------
click              8.0.4
importlib-metadata 4.8.3
libparse           0.1.5
pip                21.3.1
PyYAML             6.0.1
setuptools         39.2.0
typing_extensions  4.1.1
wheel              0.37.1
XlsxWriter         3.1.2
zipp               3.6.0

$ ls -l venv
ls: cannot access venv: No such file or directory
```

The setup sequence:

```
OL:2024.04.02 OR:0889970d1790a2617e69f253221b8bd7626e51dc dlm@host360B /data/vlsi/openlane_2024.04.02]$ prepare
Cloning into '/data/vlsi/openlane_2024.04.02'...
remote: Enumerating objects: 508, done.
remote: Counting objects: 100% (508/508), done.
remote: Compressing objects: 100% (431/431), done.
remote: Total 508 (delta 77), reused 311 (delta 32), pack-reused 0
Receiving objects: 100% (508/508), 8.33 MiB | 31.59 MiB/s, done.
Resolving deltas: 100% (77/77), done.
Note: switching to 'eaba5192c45aa333ab45216ce1773d75d539e9b3'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

###
### docker info --format {{json .}}
###
###
### docker info --format {{json .}}
###
make[1]: Entering directory `/data/vlsi/openlane_2024.04.02'
###
### docker pull efabless/openlane:eaba5192c45aa333ab45216ce1773d75d539e9b3
###
make[1]: Leaving directory `/data/vlsi/openlane_2024.04.02'
rm -rf ./venv
python3 -m venv ./venv
PYTHONPATH= ./venv/bin/python3 -m pip install --upgrade --no-cache-dir pip
Collecting pip
  Downloading https://files.pythonhosted.org/packages/a4/6d/6463d49a933f547439d6b5b98b46af8742cc03ae83543e4d7688c2420f8b/pip-21.3.1-py3-none-any.whl (1.7MB)
    100% |████████████████████████████████| 1.7MB 51.9MB/s 
Installing collected packages: pip
  Found existing installation: pip 9.0.3
    Uninstalling pip-9.0.3:
[... 260 lines cut ...]
Requires: wheel
Required-by: 
```

This is how the flow looks when it runs and has the error (outside env has libparse 0.3.1 installed, inside env does not have any libparse installed - at time it errors):

```
[STEP 31]
[INFO]: Creating IR Drop Report (log: ../work/runs/wokwi/logs/signoff/31-irdrop.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/irdrop.tcl
[ERROR]: Log: ../work/runs/wokwi/logs/signoff/31-irdrop.log
[ERROR]: Last 10 lines:
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Reading ODB at '/work/runs/wokwi/results/routing/tt_um_no_time_for_squares_tommythorn.odb'…
define_corners Typical
read_liberty -corner Typical /data/PDK_ROOT/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
Error: irdrop.tcl, 15 Traceback (most recent call last):
  File "/openlane/scripts/utils/get_default_operating_conditions.py", line 18, in <module>
    from libparse import LibertyParser
ModuleNotFoundError: No module named 'libparse'
child process exited abnormally

[ERROR]: Creating issue reproducible...
```

donn, 2024-04-06:
Addressed with #2108.

closed 2024-04-06 by donn