The-OpenROAD-Project/OpenLane #2112: Error in step-17 Running Detailed Routing

Sanchit-Gupta10, 2024-04-02:
### Description

I am geting violations in step 17(detailed routing) while running the openlane flow for my code.

[ERROR]: Last 10 lines:
    elapsed time = 00:08:25, memory = 10416.64 (MB).
    Completing 40% with 18792 violations.
    elapsed time = 00:12:11, memory = 10423.14 (MB).
    Completing 50% with 18792 violations.
    elapsed time = 00:14:06, memory = 10818.30 (MB).
    Completing 60% with 37134 violations.
    elapsed time = 00:17:14, memory = 10559.59 (MB).
    Completing 70% with 37134 violations.
    elapsed time = 00:20:59, memory = 10815.96 (MB).
child killed: kill signal

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /home/sanchit/Sanchit/Semester_2/Seminar/caravel_2d_dwt_example/caravel_user_project/openlane/user_proj_example/runs/24_04_02_11_00/issue_reproducible…
Done.

### Expected Behavior

it should run successfully

### Environment report

```shell
Kernel: Linux v6.5.0-26-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: a35b64aa200c91e9eb7dde56db787d6b4c0ea12a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a35b64a 2023-02-22T11:46:16+02:00 Checkers updates (#1665) - Kareem Farid -  (grafted, HEAD, tag: 2023.02.23)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible1.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/14832755/issue_reproducible1.zip)

### Relevant log output

```shell
[INFO DRT-0181] Start track assignment.
[INFO DRT-0184] Done with 489417 vertical wires in 6 frboxes and 305755 horizontal wires in 6 frboxes.
[INFO DRT-0186] Done with 87186 vertical wires in 6 frboxes and 105039 horizontal wires in 6 frboxes.
[INFO DRT-0182] Complete track assignment.
[INFO DRT-0267] cpu time = 00:01:15, elapsed time = 00:00:40, memory = 3620.59 (MB), peak = 4416.20 (MB)
[INFO DRT-0187] Start routing data preparation.
[INFO DRT-0267] cpu time = 00:00:00, elapsed time = 00:00:00, memory = 3620.59 (MB), peak = 4416.20 (MB)
[INFO DRT-0194] Start detail routing.
[INFO DRT-0195] Start 0th optimization iteration.
    Completing 10% with 0 violations.
    elapsed time = 00:02:49, memory = 6165.61 (MB).
    Completing 20% with 0 violations.
    elapsed time = 00:06:32, memory = 9837.24 (MB).
    Completing 30% with 18792 violations.
    elapsed time = 00:08:25, memory = 10416.64 (MB).
    Completing 40% with 18792 violations.
    elapsed time = 00:12:11, memory = 10423.14 (MB).
    Completing 50% with 18792 violations.
    elapsed time = 00:14:06, memory = 10818.30 (MB).
    Completing 60% with 37134 violations.
    elapsed time = 00:17:14, memory = 10559.59 (MB).
    Completing 70% with 37134 violations.
    elapsed time = 00:20:59, memory = 10815.96 (MB).
```

vijayank88, 2024-04-02:
@Sanchit-Gupta10 
```
elapsed time = 00:20:59, memory = 10815.96 (MB).
child killed: kill signal
```
is related to your peak memory used. How much RAM/Swap memory available?

Sanchit-Gupta10, 2024-04-02:
[image]

vijayank88, 2024-04-02:
> I dont't know how to add valid test case.. Can you please help me doing this

@Sanchit-Gupta10 
Check https://openlane.readthedocs.io/en/latest/for_developers/using_or_issue.html#manually for creating test case manually.

Sanchit-Gupta10, 2024-04-02:
@vijayank88 i have increased the swap size and now step 17 is passed successfully but now i am gettng error in the last step i.e. step 36 [Running Circuit Validity Checker ERC ]

[ERROR]: There are setup violations in the design at the typical corner.

[ERROR]: Flow failed.
[INFO]: The failure may have been because of the following warnings:
[WARNING]: There are max slew violations in the design at the typical corner. 
[WARNING]: There are max fanout violations in the design at the typical corner.
[WARNING]: There are max capacitance violations in the design at the typical corner.

vijayank88, 2024-04-02:
@Sanchit-Gupta10 
```
ERROR]: There are setup violations in the design at the typical corner.
```
Check your timing report and fix the setup violations

closed 2024-06-25 by kareefardi