The-OpenROAD-Project/OpenLane #2114: docker/*/Dockerfile has .gitignore removing yum_dependencies.txt from checkin

dlmiles, 2024-04-05:
### Description

https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docker/build_base/Dockerfile#L18

This file requires a number of *.txt files.

But a .gitignore file exists that omits them from the repository.

Other directories are similar:

```
docker$ ls -1 */.gitignore
build_base/.gitignore
openlane/.gitignore
run_base/.gitignore
tar/.gitignore

docker$ grep -n txt */.gitignore
build_base/.gitignore:1:*.txt
run_base/.gitignore:3:/*.txt
tar/.gitignore:3:/*.txt
```

It would be useful, if they can't be checked in (might break CI, security concern with exact contents), to at least provide a renamed version of each file `sample_yum_dependencies.txt`  or `yum_dependencies.txt.sample`

One issue I was looking to confirm is the centos-7 package `file` is not installed in `build-base` and some autoconf/configure use this tool when they work.

### Expected Behavior

Wanted to recreate `build-base` and `run-base`

### Environment report

```shell
N/A
```

### Reproduction material

N/A

### Relevant log output

```shell
N/A
```

donn, 2024-04-06:
This Dockerfile is not used standalone in our build process: `docker/Makefile` copies the requisite files from `dependencies`:

https://github.com/The-OpenROAD-Project/OpenLane/blob/ad7a11ff2822cf5f426da6240f1dba9c5256a7e9/docker/Makefile#L31

If you'd like to build the build and run bases for OpenLane, `make -C docker build-build-base` and `make -C docker build-run-base` should do the job.

Feel free to re-open if I misunderstood something.

closed 2024-04-06 by donn

dlmiles, 2024-04-09:
This worked great thanks.