The-OpenROAD-Project/OpenLane #2116: Automatic macro placement fails to complete PDN stage

vijayank88, 2024-04-12:
### Description

If I use automatic macro placement, flow failed during PDN stage with following error.
```
[WARNING PDN-0234] spm_inst_0 has not been placed and fixed.
[WARNING PDN-0234] spm_inst_1 has not been placed and fixed.
[ERROR PDN-0235] Design has unplaced macros.
PDN-0235
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Log: designs/ci/manual_macro_placement_test/runs/run1_amp/logs/floorplan/9-pdn.log
[ERROR]: Last 10 lines:
child process exited abnormally
```

### Expected Behavior

Complete PDN stage with automatic macro placement.

### Environment report

```shell
open_pdks cd1748bb197f9b7af62a54507de6624e30363943
Kernel: Linux v6.5.0-26-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v26.0.0 (OK)
OpenLane Git Version: ad7a11ff2822cf5f426da6240f1dba9c5256a7e9
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

ad7a11f 2024-04-03T18:41:00+02:00 Update `libparse` (#2108) - Mohamed Gaber -  (HEAD -> master, tag: 2024.04.04, origin/master, origin/HEAD)
eaba519 2024-04-02T00:07:21+02:00 Fix Exit Code Propagation When Creating Reproducibles (#2111) - Mohamed Gaber -  (tag: 2024.04.02)
c576398 2024-04-01T13:22:02+02:00 Fix Clock Port Check (#2109) - Mohamed Gaber -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/14954844/issue_reproducible.zip)

### Relevant log output

```shell
[WARNING PDN-0234] spm_inst_0 has not been placed and fixed.
[WARNING PDN-0234] spm_inst_1 has not been placed and fixed.
[ERROR PDN-0235] Design has unplaced macros.
PDN-0235
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Log: designs/ci/manual_macro_placement_test/runs/run1_amp/logs/floorplan/9-pdn.log
[ERROR]: Last 10 lines:
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /openlane/designs/ci/manual_macro_placement_test/runs/run1_amp/issue_reproducible…
Done.
[INFO]: Reproducible packaged: Please tarball and upload 'designs/ci/manual_macro_placement_test/runs/run1_amp/issue_reproducible' if you're going to submit an issue.
[ERROR]: Step 9 (floorplan) failed with error:
-code 1 -level 0 -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_tcl_script" line 219)
    invoked from within
"run_tcl_script -tool openroad -no_consume {*}$args"
    (procedure "run_openroad_script" line 2)
    invoked from within
"run_openroad_script $::env(SCRIPTS_DIR)/openroad/pdn.tcl  -indexed_log [index_file $::env(floorplan_logs)/pdn.log]  -save "to=$::env(floorplan_results..."
    (procedure "gen_pdn" line 15)
    invoked from within
"gen_pdn"
    (procedure "run_power_grid_generation" line 65)
    invoked from within
"run_power_grid_generation"
    (procedure "run_floorplan" line 56)
    invoked from within
"run_floorplan"} -errorline 1
[INFO]: Calculating runtime...
[INFO]: Saving current set of views in 'designs/ci/manual_macro_placement_test/runs/run1_amp/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/ci/manual_macro_placement_test/runs/run1_amp/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/ci/manual_macro_placement_test/runs/run1_amp/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

vijayank88, 2024-04-12:
macro placement test case added: 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/14958760/issue_reproducible.zip)

donn, 2024-04-24:
We do not support automatic macro placement in OpenLane.

closed 2024-04-24 by donn