The-OpenROAD-Project/OpenLane #2120: Clock Network Delay is 0.0 in timing reports after CTS

Jaswanth-Pappula, 2024-04-26:
Clock Network Delay is showing as 0.00 in timing reports of Max and Min after CTS step even though CTS routing is completed.

You can find it in the  image attached below.

[image]

Can someone help me in resolving this issue.

maliberty, 2024-04-26:
Most likely you need `set_propagated_clock [all_clocks]`

Jaswanth-Pappula, 2024-04-27:
> Most likely you need `set_propagated_clock [all_clocks]`

set_propagated_clock [get_clocks {Clk}] is generated in the cts.sdc file inside results section of the design but nor replicated in the min, max or nominal timing reports and it is just showing ideal.

so, is there any other way to do it instead of using set_propagated_clock [all_clocks].
[image]

donn, 2024-08-05:
Should be fixed by now.

closed 2024-08-05 by donn

Jaswanth-Pappula, 2024-08-05:
> Should be fixed by now.

Yes, the problem is fixed. Thanks for the reply