The-OpenROAD-Project/OpenLane #2122: Make test and make openlane fail

ssooffiiaannee, 2024-05-12:
### Description

While running the installation cmds:

> cd $HOME
> git clone https://github.com/The-OpenROAD-Project/OpenLane
> cd OpenLane
> make
> make test

make test leads to :
> docker: Error response from daemon: unable to find groups for spec openlane-user-1000: bufio.Scanner: token too long.
> make: *** [Makefile:204: test] Error 125

And

>  make openlane

 freezes in one build step : 
[image]

Note : make openlane, the following lines have been requested to be added to docker/build.sh

--extra-experimental-features flakes\
--extra-experimental-features nix-command\

> ++ nix build --no-link --print-out-paths --accept-flake-config --option system x86_64-linux --extra-platforms x86_64-linux ..#packages.x86_64-linux.openlane1-docker
> error: experimental Nix feature 'flakes' is disabled; add '--extra-experimental-features flakes' to enable it
> + TARBALL=

### Expected Behavior

Basic test passed

### Environment report

```shell
open_pdks bdc9412b3e468c102d01b7cf6337be06ec6e9c9a
Kernel: Linux v5.15.0-105-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.22 (OK)
OpenLane Git Version: 14b94a668289d8dc42bc15beb55ec9b3877b870c
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

14b94a66 2024-05-08T15:41:43+03:00 Migrate Build System to Match OpenLane 2 (#2115) - Mohamed Gaber -  (HEAD -> master, tag: 2024.05.09, origin/master, origin/HEAD)
337ffbf4 2024-04-21T17:01:05+02:00 Update OpenROAD (#2118) - Mohamed Gaber -  (tag: 2024.04.22)
ad7a11ff 2024-04-03T18:41:00+02:00 Update `libparse` (#2108) - Mohamed Gaber -  (tag: 2024.04.04)
---
Git Remotes

origin	git@github.com:The-OpenROAD-Project/OpenLane (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

Cloned the latest from master branch, commit : 14b94a668289d8dc42bc15beb55ec9b3877b870c

### Relevant log output

```shell
> docker: Error response from daemon: unable to find groups for spec openlane-user-1000: bufio.Scanner: token too long.
> make: *** [Makefile:204: test] Error 125
```

donn, 2024-05-12:
Substitute `make openlane` for `make pull-openlane`-- would you mind telling me where you got the installation instructions?

ssooffiiaannee, 2024-05-12:
I found it in a youtube tutorial, and after reading the Makefile. "make test" doesn't work either !

donn, 2024-05-13:
Could you perhaps link to that tutorial?

Additionally, could you tell me exactly how it failed? Keep in mind the new instructions are:

```
cd $HOME
git clone https://github.com/The-OpenROAD-Project/OpenLane
cd OpenLane
make pull-openlane
make test
```

donn, 2024-05-13:
UPDATE: appears there was a CI failure I entirely missed, and that's why it's trying to build OpenLane instead of just pulling it for you. Sincerely apologize. I will tell you when the build is ready.

donn, 2024-05-13:
Should be fine now. Please try again after a 'git pull' or two.

ssooffiiaannee, 2024-05-13:
I still get this following after pulling, this seems to be an issue inside the container, I am new to docker.
make test : 
> docker run --rm -v /home/sofiane:/home/sofiane -v /home/sofiane/OpenLane:/openlane -v /home/sofiane/OpenLane/empty:/openlane/install -v /home/sofiane/.volare:/home/sofiane/.volare -e PDK_ROOT=/home/sofiane/.volare -e PDK=sky130A  --user 1000:1002 -e DISPLAY=:0 -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/sofiane/.Xauthority:/.Xauthority --network host --security-opt seccomp=unconfined efabless/openlane:77eb5ffc00966b47b36e332ccbf42103eb95e974-amd64 sh -c "./flow.tcl -design spm -tag openlane_test -overwrite"
> docker: Error response from daemon: unable to find groups for spec openlane-user-1000: bufio.Scanner: token too long.
> make: *** [Makefile:204: test] Error 125

As for the tutorial : https://www.youtube.com/watch?v=QnJzoJjC7RQ

donn, 2024-05-14:
So by the looks of it your Docker version (`v20.10.22`) is out of date. I tried installing Ubuntu 20.04 to match your setup and everything works fine.

Did you follow this guide to install Docker? https://docs.docker.com/engine/install/ubuntu/

ssooffiiaannee, 2024-05-14:
I think so, I already had it installed long ago.
The long version of installation procedure used a version older than mine: https://openlane.readthedocs.io/en/latest/getting_started/installation/index.html. 
[image]

ssooffiiaannee, 2024-05-14:
This was a docker version issue indeed. it works with 26.1.2

donn, 2024-05-15:
That's great. I'll keep this issue open because I do need to tell people to update their Docker versions somewhere in the instructions, but thank you for the report!

mattvenn, 2024-06-25:
just found this, seems I also have to update docker

mattvenn, 2024-06-25:
updating docker fixes for me as well

PR #2132 (2024-06-26): Add docker upgrade instructions to docs
Fixes #2122
closed 2024-06-26 by donn