The-OpenROAD-Project/OpenLane #2123: Error: multi_corner.tcl line 43

fazliemre, 2024-05-20:
### Description

Hello i have get error, when i run the flow on Openlane in STEP 2.

[ERROR]: during executing sta script /openlane/scripts/openroad/sta/multi_corner.tcl
[ERROR]: Log: designs/i2c_master_top/runs/RUN_2024.05.20_19.23.01/logs/synthesis/2-sta.log
[ERROR]: Last 10 lines:
Using 1e+03 for resistance...
Using 1e-09 for time...
Using 1e+00 for voltage...
Using 1e-03 for current...
Using 1e-09 for power...
Using 1e-06 for distance...
Reading netlist '/openlane/designs/i2c_master_top/runs/RUN_2024.05.20_19.23.01/results/synthesis/i2c_master_top.v'…
Reading design constraints file at '/openlane/designs/i2c_master_top/src/i2c_master_bit_ctrl.v /openlane/designs/i2c_master_top/src/i2c_master_defines.v /openlane/designs/i2c_master_top/src/i2c_master_top.v /openlane/designs/i2c_master_top/src/timescale.v /openlane/designs/i2c_master_top/src/i2c_master_byte_ctrl.v'…
Error: multi_corner.tcl line 43, cannot open '/openlane/designs/i2c_master_top/src/i2c_master_bit_ctrl.v /openlane/designs/i2c_master_top/src/i2c_master_defines.v /openlane/designs/i2c_master_top/src/i2c_master_top.v /openlane/designs/i2c_master_top/src/timescale.v /openlane/designs/i2c_master_top/src/i2c_master_byte_ctrl.v'.
child process exited abnormally

[ERROR]: Creating issue reproducible...
[image]
[image]

### Expected Behavior

I want to pass the this step

### Environment report

```shell
OpenLane Container (2719508):/openlane$ python3 ./env.py issue-survey
open_pdks cd1748bb197f9b7af62a54507de6624e30363943
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v6.5.0-35-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 2719508edbb579ab7317d1e0942b774d8d4aa214
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

2719508e 2024-03-14T14:41:49+02:00 Remove SYNTH_READ_BLACKBOX_LIB (#2099) - Kareem Farid -  (HEAD -> master, tag: 2024.03.15, origin/master, origin/HEAD)
525d7ed3 2024-03-11T13:31:44+02:00 Defer AST Compilation in Yosys (#2100) - Kareem Farid -  (tag: 2024.03.12)
a663df28 2024-03-06T15:47:26+02:00 Update OpenROAD (#2093) - Mohamed Gaber -  (tag: 2024.03.07)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/15381145/issue_reproducible.zip)

### Relevant log output

```shell
OpenSTA 2.4.0 0889970d17 Copyright (c) 2023, Parallax Software, Inc.
License GPLv3: GNU GPL version 3 <http://gnu.org/licenses/gpl.html>

This is free software, and you are free to change and redistribute it
under certain conditions; type `show_copying' for details. 
This program comes with ABSOLUTELY NO WARRANTY; for details type `show_warranty'.
define_corners Typical
read_liberty -corner Typical /home/fazliemre/.volare/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib
Using 1e-12 for capacitance...
Using 1e+03 for resistance...
Using 1e-09 for time...
Using 1e+00 for voltage...
Using 1e-03 for current...
Using 1e-09 for power...
Using 1e-06 for distance...
Reading netlist '/openlane/designs/i2c_master_top/runs/RUN_2024.05.20_19.23.01/results/synthesis/i2c_master_top.v'…
Reading design constraints file at '/openlane/designs/i2c_master_top/src/i2c_master_bit_ctrl.v /openlane/designs/i2c_master_top/src/i2c_master_defines.v /openlane/designs/i2c_master_top/src/i2c_master_top.v /openlane/designs/i2c_master_top/src/timescale.v /openlane/designs/i2c_master_top/src/i2c_master_byte_ctrl.v'…
Error: multi_corner.tcl line 43, cannot open '/openlane/designs/i2c_master_top/src/i2c_master_bit_ctrl.v /openlane/designs/i2c_master_top/src/i2c_master_defines.v /openlane/designs/i2c_master_top/src/i2c_master_top.v /openlane/designs/i2c_master_top/src/timescale.v /openlane/designs/i2c_master_top/src/i2c_master_byte_ctrl.v'.
```

kareefardi, 2024-05-26:
@fazliemre The issue reproducible is missing some files. Can you share your design configuration file?

fazliemre, 2024-05-27:
> @fazliemre The issue reproducible is missing some files. Can you share your design configuration file?
set ::env(DESIGN_NAME) "i2c_master_top"
set ::env(PDK) "sky130A"
set ::env(VERILOG_FILES) [glob $::env(DESIGN_DIR)/src/*.v]
set ::env(CLOCK_PERIOD) 100
set ::env(CLOCK_PORT) "wb_clk_i"
set ::env(CLOCK_NET) "wb_clk_i"
set ::env(DESIGN_IS_CORE) 1
set ::env(LEC_ENABLE) 0

set ::env(RUN_LINTER) 1
set ::env(QUIT_ON_LINTER_WARNINGS) 0
set ::env(QUIT_ON_LINTER_ERRORS) 0

set ::env(SYNTH_CLOCK_UNCERTAINTY) 0.25
set ::env(SYNTH_CLOCK_UNCERTAINTY) 0.15

set ::env(SYNTH_STRATEGY) "DELAY 1"
set ::env(SYNTH_NO_FLAT) 0
set ::env(SYNTH_SHARE_RESOURCES) 1

set ::env(SYNTH_ADDER_TYPR) "YOSYS"
set ::env(BASE_SDC_FILE) [glob $::env(DESIGN_DIR)/src/*.v]
set ::env(SYNTH_FLAT_TOP) 0
set ::env(SYNTH_USE_PG_PINS_DEFINES) "USE_POWER_PINS"
set ::env(QUIT_ON_TIMING_VIOLATIONS) 1
set ::env(QUIT_ON_SETUP_VIOLATIONS) 1
set ::env(QUIT_ON_HOLD_VIOLATIONS) 1

set ::env(RUN_TAP_DECAP_INSERTION) 1
set ::env(FP_CORE_UTIL) 45
set ::env(FP_ASPECT_RATIO) 1
set ::env(FP_SIZING) "relative"
set ::env(FP_PDN_CORE_RING) 1
set ::env(FP_PDN_SKIPTRIM) 0
set ::env(VDD_NETS) [list {vccd1} {vccd2} {vdda1} {vdda2}]
set ::env(GND_NETS) [list {vssd1} {vssd2} {vssa1} {vssa2}]
set ::env(SYNTH_USE_PG_PINS_DEFINES) "USE_POWER_PINS"

set ::env(PL_TARGET_DENSITY) 55
set ::env(PL_BASIC_PLACEMENT) 0

set ::env(RUN_CTS) 1
set ::env(RUN_FILL_INSERTION) 1

set ::env(ROUTING_CORES) 4
set ::env(GRT_ALLOW_CONGESTION) 0

set ::env(RUN_CVC) 1

kareefardi, 2024-05-28:
```
set ::env(BASE_SDC_FILE) [glob $::env(DESIGN_DIR)/src/*.v]
```
@fazliemre This is not a correct configuration for `BASE_SDC_FILE`