The-OpenROAD-Project/OpenLane #2124: Yosys ABC Tool crashes with exit code 139

Fabbboy, 2024-05-22:
### Description

I encountered a segmentation fault in the ABC tool while trying to build a Verilog project within the OpenLane container. Despite attempting to resolve all warnings and even reducing the codebase to its minimal form, the error persists. This issue occurs during the synthesis stage managed by Yosys, specifically when executing ABC for technology mapping.

**I'm not sure if I have to report this here or in the Yosys Project**

### Expected Behavior

The ABC tool shouldn't crash

### Environment report

```shell
Failed to get distribution info.
open_pdks bdc9412b3e468c102d01b7cf6337be06ec6e9c9a
Kernel: Linux v5.15.146.1-microsoft-standard-WSL2
Python: v3.11.6 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found.
OpenLane Git Version: 77eb5ffc00966b47b36e332ccbf42103eb95e974
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

77eb5ff 2024-05-13T09:25:58+03:00 remove thread count from volare - Mohamed Gaber -  (grafted, HEAD -> master, tag: 2024.05.14, origin/master, origin/HEAD)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/15400932/issue_reproducible.zip)

### Relevant log output

```shell
$ ../../flow.tcl -design ./ -tag build -overwrite
[....]

[STEP 1]
[INFO]: Running Synthesis (log: runs/build/logs/synthesis/1-synthesis.log)...
[ERROR]: during executing yosys script /openlane/scripts/yosys/synth.tcl
[ERROR]: Log: runs/build/logs/synthesis/1-synthesis.log
[ERROR]: Last 10 lines:
[INFO]: USING STRATEGY AREA 0

59. Executing ABC pass (technology mapping using ABC).

59.1. Extracting gate netlist of module `\fx32' to `/tmp/yosys-abc-NMil3M/input.blif'..
Extracted 32 gates and 33 wires to a netlist network with 0 inputs and 32 outputs.

59.1.1. Executing ABC.
ERROR: ABC: execution of command ""/nix/store/5bgsgwzww9f2z7wl7kga42xrf1vpmnql-yosys-abc/bin/abc" -s -f /tmp/yosys-abc-NMil3M/abc.script 2>&1" failed: return code 139.
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...

[....]

[ERROR]: Step 1 (synthesis) failed with error:
-code 1 -level 0 -errorstack {INNER {invokeStk1 throw_error} CALL {run_tcl_script -tool yosys -no_consume /openlane/scripts/yosys/synth.tcl -indexed_log /openlane/designs/fx32/runs/build/logs/synthesis/1-synthesis.log} CALL {run_yosys_script /openlane/scripts/yosys/synth.tcl -indexed_log /openlane/designs/fx32/runs/build/logs/synthesis/1-synthesis.log} CALL {run_yosys -indexed_log /openlane/designs/fx32/runs/build/logs/synthesis/1-synthesis.log} CALL run_synthesis CALL {run_non_interactive_mode -design ./ -tag build -overwrite}} -errorcode NONE -errorinfo {
    while executing
"throw_error"
    (procedure "run_tcl_script" line 219)
    invoked from within
"run_tcl_script -tool yosys -no_consume {*}$args"
    (procedure "run_yosys_script" line 2)
    invoked from within
"run_yosys_script $::env(SYNTH_SCRIPT) -indexed_log $arg_values(-indexed_log)"
    (procedure "run_yosys" line 40)
    invoked from within
"run_yosys -indexed_log $log"
    (procedure "run_synthesis" line 13)
    invoked from within
"run_synthesis"} -errorline 1
[INFO]: Saving current set of views in 'runs/build/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'runs/build/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'runs/build/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

kareefardi, 2024-05-26:
@Fabbboy The issue reproducible is missing some files. Since this crash happens early in the flow, can you upload your input files (config, rtl, gds..)? It will also be useful to determine if there is an issue in producing the reproducibles in OpenLane.

Fabbboy, 2024-05-28:
This is weird. I just zipped the folder and did not change anything. Sadly I currently have no more access to the project.... This was kinda dumb. I give my best to recover it.

Fabbboy, 2024-05-29:
@kareefardi 
I'm not able to upload the project because it's too big. How should I provide it to you?

Fabbboy, 2024-05-29:
[reports.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/15481066/reports.zip)
[results.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/15481067/results.zip)
[logs.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/15481068/logs.zip)

kareefardi, 2024-06-23:
@Fabbboy I can confirm the issue. I will try compiling the latest yosys and see if the crash persists.

Fabbboy, 2024-06-23:
Perfect I'm glad its not my fault

kareefardi, 2024-07-02:
Created a test case and submitted an issue to Yosys https://github.com/YosysHQ/yosys/issues/4473