The-OpenROAD-Project/OpenLane #2129: Issue with installation of OpenLane, while running make cmd "Error: Command '['/home/HP/OpenLane/venv/bin/python3', '-Im', 'ensurepip', '--upgrade', '--default-pip']' returned non-zero exit status 1."

WrathofBhuvan11, 2024-06-21:
[image]

Here one can see when I run the make command for installation, it throws an error with Python. Please help me with resolving the issue.

donn, 2024-06-25:
Are you following the installation guide in https://openlane.readthedocs.io/en/latest/getting_started/installation/index.html ?

WrathofBhuvan11, 2024-07-02:
Yes, I'm using Cygwin-X Terminal which is integrated with Windows python (Anaconda installed). I'm running this in Cygwin-X

WrathofBhuvan11, 2024-07-16:
Please someone reply

donn, 2024-07-17:
We're leaving it open so someone in the community could maybe reply but we don't have much experience with neither Cygwin nor Conda…

WrathofBhuvan11, 2024-07-30:
> We're leaving it open so someone in the community could maybe reply but we don't have much experience with neither Cygwin nor Conda… 

Actually I'm currently trying in Ubuntu env windows. Cygwin is just that path but I'm using Ubuntu Linux WSL to run these.

The last reply screenshot is from Ubuntu WSL

kareefardi, 2024-08-01:
@WrathofBhuvan11 did you update and install required packages in WSL?
```
sudo apt-get update
sudo apt-get upgrade
sudo apt install -y build-essential python3 python3-venv make git
```

donn, 2024-09-12:
I'm afraid we cannot support a conda installation, period. Too many variables.

closed 2024-09-12 by donn