The-OpenROAD-Project/OpenLane #2130: [ERROR GRT-0118] Routing congestion too high.

Sanchit-Gupta10, 2024-06-24:
### Description

i am running openlane flow but it got stuck at global routing step 
i have also attached the config.json file.

### Expected Behavior

the flow should be completed

### Environment report

```shell
Kernel: Linux v6.5.0-35-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.5 (OK)
OpenLane Git Version: a35b64aa200c91e9eb7dde56db787d6b4c0ea12a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: FAILED
/home/sanchit/Sanchit/Semester_2/Seminar/caravel_user_project/dependencies/openlane_src/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/sanchit/Sanchit/Semester_2/Seminar/caravel_user_project/dependencies/openlane_src/dependencies/verify_versions.py", line 76, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/sanchit/Sanchit/Semester_2/Seminar/caravel_user_project/dependencies/openlane_src/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

a35b64a 2023-02-22T11:46:16+02:00 Checkers updates (#1665) - Kareem Farid -  (grafted, HEAD, tag: 2023.02.23)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
```

### Reproduction material

[issue_reproducible.zip](https://github.com/user-attachments/files/15952287/issue_reproducible.zip)
i have also added config.json file for reference
[config.json](https://github.com/user-attachments/files/15952319/config.json)

### Relevant log output

```shell
[STEP 16]
[INFO]: Running Global Routing (log: ../home/caravel_user_project/openlane/user_proj_example/runs/24_06_24_13_46/logs/routing/16-global.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/groute.tcl
[ERROR]: Log: ../home/caravel_user_project/openlane/user_proj_example/runs/24_06_24_13_46/logs/routing/16-global.log
[ERROR]: Last 10 lines:
[INFO GRT-0101] Running extra iterations to remove overflow.
[WARNING GRT-0227] Reached 20 congestion iterations with less than 15% of reduction between iterations.
[INFO GRT-0197] Via related to pin nodes: 277576
[INFO GRT-0198] Via related Steiner nodes: 19441
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 393363
[INFO GRT-0112] Final usage 3D: 1859063
[ERROR GRT-0118] Routing congestion too high. Check the congestion heatmap in the GUI.
Error: groute.tcl, 36 GRT-0118
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager
```

vijayank88, 2024-06-24:
```
#unzip issue_reproducible.zip 
Archive:  issue_reproducible.zip
warning [issue_reproducible.zip]:  zipfile claims to be last disk of a multi-part archive;
  attempting to process anyway, assuming all parts have been concatenated
  together in order.  Expect "errors" and warnings...true multi-part support
  doesn't exist yet (coming soon).
```
zip file missing some more files. Attach a valid reproducible test case.

Sanchit-Gupta10, 2024-06-24:
[tmp.zip](https://github.com/user-attachments/files/15952740/tmp.zip)
[issue_reproducible_1.zip](https://github.com/user-attachments/files/15952742/issue_reproducible_1.zip)
i have separated the temp folder from issue reproducible

Sanchit-Gupta10, 2024-06-24:
@vijayank88 i have decreased the diode_insertion_strategy value from 4 to 3 and now the openlane flow is completed successfully.
But, i didn't know exactly that whether it is correct way of handling this error or not?

maliberty, 2024-06-24:
Does doing so result in fixing all antenna violations?

vijayank88, 2024-06-24:
@Sanchit-Gupta10 
Seems you're using last year OpenLane commit `2023.02.23`. can you update your OpenLane to latest and try once.

Sanchit-Gupta10, 2024-06-24:
> Does doing so result in fixing all antenna violations?
@maliberty No, it shows that
[INFO] [Iteration 1] Failed to reduce antenna violations (54->54), stopping iterations

but it doesn't give any errors.

Sanchit-Gupta10, 2024-06-24:
> @Sanchit-Gupta10 Seems you're using last year OpenLane commit `2023.02.23`. can you update your OpenLane to latest and try once.

okay i will try with latest version

Sanchit-Gupta10, 2024-06-24:
@maliberty @vijayank88 during routing step this is showing
[INFO] [Iteration 1] Failed to reduce antenna violations (54->54), stopping iterations

there is no error in the openlane flow so, can i proceed futher ??

vijayank88, 2024-06-24:
` Failed to reduce antenna violations (54->54), stopping iterations` it means antenna violations not fixed completely.

kareefardi, 2024-06-25:
@Sanchit-Gupta10 Check antenna report file inside `reports/signoff`

closed 2024-09-12 by donn