The-OpenROAD-Project/OpenLane #2140: Building PDK and Tested PDK mismatch

ihaziqiqbal, 2024-07-16:
Hi. I have just installed OpenLane and SKY130 PDK on my Fedora machine and ran the test script fine. Now I am trying to initialize my very first basic inverter design. So I made a new folder "inverter_try" in OpenLane/designs directory and  

make mount
./flow.tcl -design inverter_try -init_design_config

and received following error:

```
> [ERROR]: open_pdks 06898d0576a1820a131f58b05b6af5f504f080d9
> magic e9db9ecbc9943a304de80b32ecc58a61a46cc91f
> Failed to compare PDKs.
> The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: 06898d0576a1820a131f58b05b6af5f504f080d9, tested: bdc9412b3e468c102d01b7cf6337be06ec6e9c9a)
> This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
> 'magic'
> Traceback (most recent call last):
>   File "/openlane/dependencies/verify_versions.py", line 132, in verify_versions
>     manifest_commit = manifest_dict[name]["commit"]
>                       ~~~~~~~~~~~~~^^^^^^
> KeyError: 'magic'
> 
> [ERROR]: Please update your environment. OpenLane will now quit.
```

So I exited the docker and exexuted make pdk once again and it gave the following result

`Version bdc9412b3e468c102d01b7cf6337be06ec6e9c9a enabled for the sky130 PDK.`

But I am still receiving same pdk mismatch error. Any help is greatly appreciated.

kareefardi, 2024-07-28:
@ihaziqiqbal how did you build OpenLane and how did you enter the run the docker container?

ihaziqiqbal, 2024-07-28:
Here's how I did it.

```
git clone --depth 1 https://github.com/The-OpenROAD-Project/OpenLane.git
cd OpenLane/
make
make test
```

Basically, I followed everything from here.
https://openlane.readthedocs.io/en/latest/getting_started/installation/installation_linux.html

For the docker, followed this page.
https://docs.docker.com/engine/install/fedora/

And for sky130 pdk.

```
./configure --enable-sky130-pdk --enable-sram-sky130
make
sudo make install
make veryclean
```

kareefardi, 2024-08-01:
@ihaziqiqbal `make` in OpenLane should have installed the PDK and no need to clone sky130 pdk and install it standalone. This would actually lead to the mismatch that you reported since the version installed by `make` in OpenLane is not always the latest version available by open_pdks

kareefardi, 2024-08-01:
having said, I am wondering if you are setting `PDK_ROOT` to a different value.

ihaziqiqbal, 2024-08-01:
@kareefardi when I did `make` Openlane, I couldn't find any any directory where sky130A was located so I thought it needs to be installed separately.

Also when I did `make mount` in OpenLane directory, it showed `PDK_ROOT=/home/haziqiqbal/.volare` which I was not sure what it is becasuse ./volare it was not in the directory. So I installed pdk and set `PDK_ROOT=/usr/local/share/pdk/`

donn, 2024-08-14:
Kindly submit a new issue following the issue template. The information in it will help us solve your issue faster.

closed 2024-08-14 by donn