The-OpenROAD-Project/OpenLane #2143: Version Mismatch and KeyError for 'magic' when Switching to GF180mcu

gowreddysuresh, 2024-08-01:
### Description

I encountered an error when trying to switch the technology node from SKY130nm to GF180mcu in OpenLane. Despite changing the configuration file, the tool fails to run and shows a version mismatch error and a KeyError for 'magic'

### Expected Behavior

The tool should recognize the GF180mcu technology node after the configuration change and proceed without any version mismatch errors or KeyErrors. It should successfully run the flow for the specified design

### Environment report

```shell
CentOS Linux release 7.9.2009 (Core)
```

### Reproduction material

Changed the technology node path in Makefile from SKY130nm to GF180mcu.
2. Ran the following command:
   ```sh
   make mount
   cd /openlane
   ./flow.tcl -design fa

### Relevant log output

```shell
[ERROR]: open_pdks 06898d0576a1820a131f58b05b6af5f504f080d9
magic 81b5ac20797d64d262b905e324cc401d01f58d57
Failed to compare PDKs.
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: 06898d0576a1820a131f58b05b6af5f504f080d9, tested: bdc9412b3e468c102d01b7cf6337be06ec6e9c9a)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
'magic'
Traceback (most recent call last):
  File "/openlane/dependencies/verify_versions.py", line 132, in verify_versions
    manifest_commit = manifest_dict[name]["commit"]
                      ~~~~~~~~~~~~~^^^^^^
KeyError: 'magic'

[ERROR]: Please update your environment. OpenLane will now quit.
```

proppy, 2024-08-09:
```
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: 06898d0576a1820a131f58b05b6af5f504f080d9, tested: bdc9412b3e468c102d01b7cf6337be06ec6e9c9a)
```
can you share how you installed the PDK?

gowreddysuresh, 2024-08-09:
# Get the open_pdks installer and build Sky130 and GF180MCU
echo "Get the open_pdks installer and build Sky130 and GF180MCU"
git clone https://github.com/RTimothyEdwards/open_pdks
cd open_pdks
./configure --enable-sky130-pdk --enable-sram-sky130
make
sudo make install
make veryclean
./configure --enable-gf180mcu-pdk --enable-osu-sc-gf180mcu
make
sudo make install
make veryclean
make distclean
cd ~/gits

donn, 2024-08-14:
Is there any particular reason you didn't run the environment report script in the issue template?

donn, 2024-12-22:
Staled out.

closed 2024-12-22 by donn