The-OpenROAD-Project/OpenLane #2144: getting max capacitance and max slew violations at some corners while doing sta

Sanchit-Gupta10, 2024-08-02:
### Description

i have introduced some buffers in order to reduce the capacitance violations but these are still some violations left.
please suggest me something to remove these violations.
[image]
When i am hardening my user_proj_example and user_project_wrapper then i got warnings also about slew/cap/fanout violations but the flow was completed so i did not look into it.

### Expected Behavior

no violations should be there.

### Environment report

```shell
Kernel: Linux v6.5.0-44-generic
Distribution: ubuntu 22.04
Python: v3.10.12 (OK)
Container Engine: docker v24.0.7 (OK)
OpenLane Git Version: a35b64aa200c91e9eb7dde56db787d6b4c0ea12a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: FAILED
/home/sanchit/Sanchit/Semester_2/Seminar/caravel_user_project/dependencies/openlane_src/pdks/sky130A not found.
Traceback (most recent call last):
  File "/home/sanchit/Sanchit/Semester_2/Seminar/caravel_user_project/dependencies/openlane_src/dependencies/verify_versions.py", line 76, in verify_versions
    raise Exception(f"{pdk_dir} not found.")
Exception: /home/sanchit/Sanchit/Semester_2/Seminar/caravel_user_project/dependencies/openlane_src/pdks/sky130A not found.

Failed to verify sky130A.
---
Git Log (Last 3 Commits)

a35b64a 2023-02-22T11:46:16+02:00 Checkers updates (#1665) - Kareem Farid -  (grafted, HEAD, tag: 2023.02.23)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

no issue 
i have included config.json file of user_proj_example 
[config.json](https://github.com/user-attachments/files/16470488/config.json)

### Relevant log output

```shell
=============================================Summary=============================================
 corner|   min delay |  min reg2reg |    max delay |  max reg2reg |      max cap |     max slew
  f-max|  vio(-0.01) |   vio(-0.01) |          met |          met |          met |          met
  f-min|         met |          met |          met |          met |          met |          met
  f-nom|         met |          met |          met |          met |          met |          met
  s-max|         met |          met |          met |          met |   vio(-0.12) |   vio(-1.74)
  s-min|         met |          met |          met |          met |   vio(-0.08) |   vio(-1.24)
  s-nom|         met |          met |          met |          met |   vio(-0.10) |   vio(-1.52)
  t-max|         met |          met |          met |          met |   vio(-0.06) |   vio(-0.47)
  t-min|         met |          met |          met |          met |   vio(-0.02) |   vio(-0.18)
  t-nom|         met |          met |          met |          met |   vio(-0.04) |   vio(-0.35)
=================================================================================================
```

Sanchit-Gupta10, 2024-08-02:
@vijayank88 some suggestions please

donn, 2024-08-14:
You could decrease the max capacitance violations by setting a more aggressive fanout constraint early on (during synthesis): for example, set it to `6` instead of the default `10`. This may also solve your slew violations. However, it will naturally consume more area.

---

It is sometimes possible to waive max slew violations (at your own risk.) Absent hold and setup violations, max cap/slew violations really only affect the accuracy of the timing data, as timing values for slews higher than 1.5 in sky130_fd_sc_hd would have to be extrapolated.

You need to keep these points in mind:

* The magnitude of the violation: In some corners, it's a relatively low-risk `-0.35`, but in others it's a very high `-1.74`.
* Some of these max slew violations you showed us are on diodes. These don't matter at all and can be waived.
* Check these cells and make sure they're not part of the clock tree. If they are part of the clock tree, you must not waive them as these could potentially cause hold violations.

Sanchit-Gupta10, 2024-08-19:
[image]

donn, 2024-09-12:
You can decrease it further. But you have hold violations in the fast corner which is a much bigger problem.