The-OpenROAD-Project/OpenLane #2158: Upgrade to rtl_macro_placer

maliberty, 2024-10-02:
### Description

- OL is still using the old macro placer (macro_placement) rather than the new one (rtl_macro_placer).   The old one should be considered deprecated and will eventually be removed from OR.

### Proposal

Switch to rtl_macro_placer.  https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/scripts/macro_place_util.tcl shows its usage in ORFS.

donn, 2024-10-02:
Acknowledged- will decide on the proper course of action since OpenLane 1's whole raison d'être now is that we don't change things too much

We'll definitely do that for 2 though

maliberty, 2024-10-02:
Does that mean you don't update OR in OL1?  If so then there is no constraint from it.

donn, 2024-10-02:
Yeah, we don't unless we have a very good reason