The-OpenROAD-Project/OpenLane #2159: issue running the make file

claudiotalarico, 2024-10-04:
### Description

On Sept 27. I installed openlane and went through the instructions posted on the git-hub:
```
cd $HOME
git clone https://github.com/The-OpenROAD-Project/OpenLane OpenLane_git
cd OpenLane
make
make test
```

Note that for no good reason I decided to call my cloned directory OpenLabe_git (rather than just OpenLane)
Everything runs as expected.

Today (Oct 4, 2024) I repeated the exact same steps on the same machine (calling the cloned directory OpenLane to do do not overwrite the old one).
```
cd $HOME
git clone https://github.com/The-OpenROAD-Project/OpenLane
cd OpenLane
make
make test
```
When running make the "process" exits with ERRORS

If I cd in OpenLane_git and run make and make test everything still works fine (from within the old repository)
There seem to be changes in the github that create an issue, but I am not able to identify them.

### Expected Behavior

I expect the installation to be successfull as specified in the documentation.

### Environment report

```shell
Kernel: Darwin v23.6.0
Distribution: macOS 14.6.1
Python: v3.11.5 (OK)
Container Engine: docker v24.0.6 (OK)
OpenLane Git Version: 1b3cf10884bfae06d8fedceb0d2debd97ea351dd
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

1b3cf10 2024-10-02T11:10:31+03:00 bugfix: Replaced inappropriate `is` comparison with `==` (#2157) - Ataf Fazledin Ahamed -  (grafted, HEAD -> master, origin/master, origin/HEAD)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

I did not add any file. I am just testing the installation with the defaults.

### Relevant log output

```shell
(base) talarico@Claudios-Mac-Studio ~ % cd ~/OpenLane
(base) talarico@Claudios-Mac-Studio OpenLane % make
gmake[1]: Entering directory '/Users/talarico/OpenLane'
Error response from daemon: manifest for efabless/openlane:1b3cf10884bfae06d8fedceb0d2debd97ea351dd not found: manifest unknown: manifest unknown
gmake[1]: *** [Makefile:125: pull-openlane] Error 1
gmake[1]: Leaving directory '/Users/talarico/OpenLane'
gmake[1]: Entering directory '/Users/talarico/OpenLane'
gmake[2]: Entering directory '/Users/talarico/OpenLane/docker'
NIX_SYSTEM=aarch64-linux BUILD_ARCH=arm64v8 bash build.sh
+ set -e
++ nix build --no-link --print-out-paths --accept-flake-config --option system aarch64-linux --extra-platforms aarch64-linux ..#packages.aarch64-linux.openlane1-docker
warning: ignoring the client-specified setting 'extra-platforms', because it is a restricted setting and you are not a trusted user
warning: ignoring the client-specified setting 'system', because it is a restricted setting and you are not a trusted user
warning: ignoring the client-specified setting 'extra-platforms', because it is a restricted setting and you are not a trusted user
warning: ignoring the client-specified setting 'system', because it is a restricted setting and you are not a trusted user
warning: ignoring the client-specified setting 'trusted-public-keys', because it is a restricted setting and you are not a trusted user
error: a 'aarch64-linux' with features {} is required to build '/nix/store/y6f22zgccxk1ay8ygfisq3rps3d272d5-channel-nixos.drv', but I am a 'aarch64-darwin' with features {apple-virt, benchmark, big-parallel, nixos-test}
+ TARBALL=
gmake[2]: *** [Makefile:13: openlane] Error 1
gmake[2]: Leaving directory '/Users/talarico/OpenLane/docker'
gmake[1]: *** [Makefile:117: openlane] Error 2
gmake[1]: Leaving directory '/Users/talarico/OpenLane'
gmake: *** [Makefile:128: get-openlane] Error 2
```

donn, 2024-10-04:
I think the latest commit hasn't been pushed to Dockerhub, so your computer attempted to build it. As you use macOS, your installation of Nix cannot build the Linux-based Docker image.

The latest commit not being pushed to Dockerhub is a huge problem. I will address that immediately. Thank you for the bug report.

donn, 2024-10-04:
I manually just pushed the requisite images, so `make` should work for you now. I'll be addressing the root cause momentarily.

claudiotalarico, 2024-10-04:
Thanks a lot,
Claudio

donn, 2024-10-05:
Fixed in 3876562d27af3f6825a823941b1cab36f7eb6dc3. Thank you!

closed 2024-10-05 by donn