The-OpenROAD-Project/OpenLane #2160: Issue with OpenLane 1 and with Klayout and starting correctly from within "make mount"

ctalar, 2024-10-05:
### Description

On MacOS BigSur (Processor Intel: i7)
after successfully running:
```
cd $HOME
cd OpenLane
make
make test
make mount
```
When invoking klayout to see the gds generated, klayout doesn't start  as expected:  
`klayout ./designs/spm/runs/openlane_test/results/final/gds/spm.gds`

```
OpenLane Container (1.1.1):/openlane% klayout -v
KLayout 0.29.1
```

### Expected Behavior

Expect klayout to start and visualize the generated gds

### Environment report

```shell
(base) Claudios-MacBook-Pro:OpenLane claudio$ python3 ./env.py issue-survey
Kernel: Darwin v20.6.0
Distribution: macOS 10.16
Python: v3.11.5 (OK)
Container Engine: docker v24.0.6 (OK)
OpenLane Git Version: 3876562d27af3f6825a823941b1cab36f7eb6dc3
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

3876562d 2024-10-05T02:47:02+03:00 hotfix: main branch hashes not being pushed to dockerhub - Mohamed Gaber -  (HEAD -> master, origin/master, origin/HEAD)
1b3cf108 2024-10-02T11:10:31+03:00 bugfix: Replaced inappropriate `is` comparison with `==` (#2157) - Ataf Fazledin Ahamed -  ()
70581ea9 2024-09-22T13:27:10+03:00 Change Versioning Strategy (master) (#2154) - Mohamed Gaber -  (tag: 1.1.0)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

no files used

### Relevant log output

```shell
klayout ./designs/spm/runs/openlane_test/results/final/gds/spm.gds
Info: Could not load the Qt platform plugin "xcb" in "" even though it was found.
Fatal: This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

zsh: IOT instruction  klayout ./designs/spm/runs/openlane_test/results/final/gds/spm.gds
```

donn, 2024-10-05:
OpenLane using Docker does not support GUIs on macOS, sadly. You'd have to modify the Makefile and use [XQuartz](https://www.xquartz.org) to do that, which is very finicky and highly setup-dependent. (Heck, even on Linux it's something of a nightmare situation now what with SELinux, AppArmor and similar…)

If you'd like a GUI on macOS, the official supported solution is installing OpenLane 2.0 or later using Nix as shown in this guide: https://openlane2.readthedocs.io/en/latest/getting_started/common/nix_installation/installation_macos.html

closed 2024-10-05 by donn

ctalar, 2024-10-06:
Thanks Mohamed, that’s very good to know.
I was starting to think I was jinxed :-)

I guess it is not a big deal, one can install klayout externally to see the gds.

Claudio

Closed #2160<https://github.com/The-OpenROAD-Project/OpenLane/issues/2160> as not planned.