The-OpenROAD-Project/OpenLane #2162: Errors found by linter

Thanhdat1301, 2024-10-22:
### Description

I am trying to add a verilog file to OpenLane. I followed the instructions in getting_started. Here I create a code counter file then run the following command ./flow.tcl -design <design_name>. However it gives the following error. Does anyone have a fix for this?
[image]

kareefardi, 2024-10-22:
Can you upload your design and configuration files?

kareefardi, 2024-10-22:
Also please follow the issue template and add the information from 
```
python3 ./env.py issue-survey
```

Thanhdat1301, 2024-10-22:
> Can you upload your design and configuration files?
I create a simple verilog counter file as follows:
// 4-bit counter module in Verilog
module counter (
    input wire clk,       // Clock input
    input wire rst_n,     // Active low reset input
    output reg [3:0] count // 4-bit counter output
);

always @(posedge clk or negedge rst_n) begin
    if (!rst_n)
        count <= 4'b0000; // Reset counter to 0
    else
        count <= count + 1; // Increment counter on clock edge
end

endmodule
This is config.json, I haven't changed anything in the config file, it generates it automatically.
	
DESIGN_NAME	"Demo"
VERILOG_FILES	"dir::src/*.v"
CLOCK_PORT	"clk"
CLOCK_PERIOD	10
FP_PDN_MULTILAYER	true

Thanhdat1301, 2024-10-22:
I tested a simple counter file:
// 4-bit counter module in Verilog
module counter (
    input wire clk,       // Clock input
    input wire rst_n,     // Active low reset input
    output reg [3:0] count // 4-bit counter output
);

always @(posedge clk or negedge rst_n) begin
    if (!rst_n)
        count <= 4'b0000; // Reset counter to 0
    else
        count <= count + 1; // Increment counter on clock edge
end

endmodule
I haven't changed anything in this config file, it's generated automatically from the commands I followed in getting_started
	
DESIGN_NAME	"Demo"
VERILOG_FILES	"dir::src/*.v"
CLOCK_PORT	"clk"
CLOCK_PERIOD	10
FP_PDN_MULTILAYER	true

d-mitch-bailey, 2024-10-22:
@Thanhdat1301 the design name is `Demo`, but the only module I see in the verilog is `counter`.
Try changing the design name to `counter` or the verilog module to `Demo`

d-mitch-bailey, 2024-10-22:
@Thanhdat1301 Now the error message is `Pitch x is too small`. Try a fixed design size with 
```
    "FP_SIZING": "absolute",
    "DIE_AREA": "0 0 1000 1000",
```

Thanhdat1301, 2024-10-23:
@d-mitch-bailey  Can I ask where I can change the design size?

d-mitch-bailey, 2024-10-23:
@Thanhdat1301 each design is synthesized and created using the `openlane/<design>/config.json` or `openlane/<design>/config.json` files. I don't think these are automatically generated.

donn, 2024-12-22:
1. Please adhere to the issue template. It is incredibly difficult to help people when we lack the basic knowledge as to what they use.
2. Please start a new issue if you're experiencing a new problem.

closed 2024-12-22 by donn