The-OpenROAD-Project/OpenLane #2167: Warning PDC_FILE

srknkrdnz, 2024-11-15:
### Description

This is my config.tcl file which is not running properly 
[image]
[image]

### Expected Behavior

I got this report How can ı fix 
[image]
[image]
[image]

### Environment report

```shell
x
```

### Reproduction material

z

### Relevant log output

```shell
z
```

srknkrdnz, 2024-11-15:
[image]

vijayan-krishna, 2024-11-16:
@srknkrdnz 
Problem not related to warning... Real error in logs/synthesis/linter.log
Check the log and fix it.

srknkrdnz, 2024-11-16:
Thanks for recommendation. I found error.

closed 2024-11-16 by srknkrdnz