The-OpenROAD-Project/OpenLane #2177: How can I remove the annoying synthesis step???

redpanda3, 2024-12-19:
### Description

Dear devs,

How can I remove the synthesis step??? The IP to use is not so synthesizable by yosys, so I tried some commercial flow and switched back to openlane for hint. When I use the -from floorplan, it says that this is no longer supported, I add     "SYNTH_READ_BLACKBOX_LIB": 1,
    "SYNTH_ELABORATE_ONLY": 1, does no work.

Best,
redpanda or yuda

### Expected Behavior

Truely, I want everything to go back, please, do not upgrade so fast...

### Environment report

```shell
NA
```

### Reproduction material

NA

### Relevant log output

```shell
NA
```

d-m-bailey, 2024-12-19:
@redpanda3 Can you share your `config.json` file?

redpanda3, 2024-12-19:
{
    "DESIGN_NAME": "xxxxx",
    "DESIGN_IS_CORE": 0,
    "VERILOG_FILES": [
        "dir::../../verilog/rtl/xxxxx"
    ],
    "CLOCK_PORT": "clk",
    "CLOCK_NET": "clk",
    "RUN_HEURISTIC_DIODE_INSERTION": true,
    "SKIP_SYNTHESIS":true,
    "NO_SYNTH_CELL_LIST": "dir::no_synth.cells",
    "SYNTH_READ_BLACKBOX_LIB": 1,
    "SYNTH_ELABORATE_ONLY": 1,
    "SYNTH_FLAT_TOP": true,
    "DIE_AREA": "0 0 2800 1760",
    "DIODE_ON_PORTS": "in",
    "FP_CORE_UTIL": 15,
    "pdk::sky130*": {
        "scl::sky130_fd_sc_hd": {
            "MAX_FANOUT_CONSTRAINT": 8,
            "PL_TARGET_DENSITY": 0.25,
            "PL_RESIZER_HOLD_SLACK_MARGIN": 0.25,
            "CLOCK_PERIOD": 25
        },
        "scl::sky130_fd_sc_hdll": {
            "MAX_FANOUT_CONSTRAINT": 8,
            "PL_TARGET_DENSITY": 0.25,
            "CLOCK_PERIOD": 25.9
        },
        "scl::sky130_fd_sc_hs": {
            "MAX_FANOUT_CONSTRAINT": 8,
            "PL_TARGET_DENSITY": 0.25,
            "CLOCK_PERIOD": 25.9
        },
        "scl::sky130_fd_sc_ls": {
            "MAX_FANOUT_CONSTRAINT": 8,
            "PL_TARGET_DENSITY": 0.25,
            "CLOCK_PERIOD": 25.9
        },
        "scl::sky130_fd_sc_ms": {
            "MAX_FANOUT_CONSTRAINT": 8,
            "PL_TARGET_DENSITY": 0.25,
            "CLOCK_PERIOD": 25.9
        }
    }
}

redpanda3, 2024-12-19:
@d-m-bailey Sure

d-m-bailey, 2024-12-19:
@redpanda3 Can I suggest using the default `caravel_user_project/openlane/user_project_wrapper/config.json` file as a starting point? This design configuration performs no synthesis.
```
{
    "DESIGN_NAME": "user_project_wrapper",
    "VERILOG_FILES": [
        "dir::../../verilog/rtl/defines.v",
        "dir::../../verilog/rtl/user_project_wrapper.v"
    ],
    "ROUTING_CORES": 1,
    "CLOCK_PERIOD": 25,
    "CLOCK_PORT": "wb_clk_i",
    "CLOCK_NET": "mprj.clk",
    "FP_PDN_MACRO_HOOKS": "mprj vccd1 vssd1 vccd1 vssd1",
    "MACRO_PLACEMENT_CFG": "dir::macro.cfg",
    "MAGIC_DEF_LABELS": 0,
    "VERILOG_FILES_BLACKBOX": [
        "dir::../../verilog/gl/user_proj_example.v"
    ],
    "EXTRA_LEFS": "dir::../../lef/user_proj_example.lef",
    "EXTRA_GDS_FILES": "dir::../../gds/user_proj_example.gds",
    "EXTRA_LIBS": "dir::../../lib/user_proj_example.lib",
    "EXTRA_SPEFS": [
        "user_proj_example", 
        "dir::../../spef/multicorner/user_proj_example.min.spef", 
        "dir::../../spef/multicorner/user_proj_example.nom.spef", 
        "dir::../../spef/multicorner/user_proj_example.max.spef"
    ],
    "BASE_SDC_FILE": "dir::base_user_project_wrapper.sdc",
    "IO_SYNC": 0,
    "MAX_TRANSITION_CONSTRAINT": 1.5,
    "RUN_LINTER": 0,
    "QUIT_ON_SYNTH_CHECKS": 0,
    "FP_PDN_CHECK_NODES": 0,
    "SYNTH_ELABORATE_ONLY": 1,
    "PL_RANDOM_GLB_PLACEMENT": 1,
    "PL_RESIZER_DESIGN_OPTIMIZATIONS": 0,
    "PL_RESIZER_TIMING_OPTIMIZATIONS": 0,
    "GLB_RESIZER_DESIGN_OPTIMIZATIONS": 0,
    "GLB_RESIZER_TIMING_OPTIMIZATIONS": 0,
    "PL_RESIZER_BUFFER_INPUT_PORTS": 0,
    "FP_PDN_ENABLE_RAILS": 0,
    "GRT_REPAIR_ANTENNAS": 0,
    "RUN_FILL_INSERTION": 0,
    "RUN_TAP_DECAP_INSERTION": 0,
    "FP_PDN_VPITCH": 180,
    "FP_PDN_HPITCH": 180,
    "RUN_CTS": 0,
    "FP_PDN_VOFFSET": 5,
    "FP_PDN_HOFFSET": 5,
    "MAGIC_ZEROIZE_ORIGIN": 0,
    "FP_SIZING": "absolute",
    "RUN_CVC": 0,
    "UNIT": 2.4,
    "FP_IO_VEXTEND": "expr::2 * $UNIT",
    "FP_IO_HEXTEND": "expr::2 * $UNIT",
    "FP_IO_VLENGTH": "expr::$UNIT",
    "FP_IO_HLENGTH": "expr::$UNIT",
    "FP_IO_VTHICKNESS_MULT": 4,
    "FP_IO_HTHICKNESS_MULT": 4,
    "FP_PDN_CORE_RING": 1,
    "FP_PDN_CORE_RING_VWIDTH": 3.1,
    "FP_PDN_CORE_RING_HWIDTH": 3.1,
    "FP_PDN_CORE_RING_VOFFSET": 12.45,
    "FP_PDN_CORE_RING_HOFFSET": 12.45,
    "FP_PDN_CORE_RING_VSPACING": 1.7,
    "FP_PDN_CORE_RING_HSPACING": 1.7,
    "FP_PDN_VWIDTH": 3.1,
    "FP_PDN_HWIDTH": 3.1,
    "FP_PDN_VSPACING": "expr::(5 * $FP_PDN_CORE_RING_VWIDTH)",
    "FP_PDN_HSPACING": "expr::(5 * $FP_PDN_CORE_RING_HWIDTH)",
    "VDD_NETS": [
        "vccd1",
        "vccd2",
        "vdda1",
        "vdda2"
    ],
    "GND_NETS": [
        "vssd1",
        "vssd2",
        "vssa1",
        "vssa2"
    ],
    "SYNTH_USE_PG_PINS_DEFINES": "USE_POWER_PINS",
    "pdk::sky130*": {
        "RT_MAX_LAYER": "met4",
        "DIE_AREA": "0 0 2920 3520",
        "FP_DEF_TEMPLATE": "dir::fixed_dont_change/user_project_wrapper.def",
        "scl::sky130_fd_sc_hd": {
            "CLOCK_PERIOD": 25
        },
        "scl::sky130_fd_sc_hdll": {
            "CLOCK_PERIOD": 10
        },
        "scl::sky130_fd_sc_hs": {
            "CLOCK_PERIOD": 8
        },
        "scl::sky130_fd_sc_ls": {
            "CLOCK_PERIOD": 10,
            "SYNTH_MAX_FANOUT": 5
        },
        "scl::sky130_fd_sc_ms": {
            "CLOCK_PERIOD": 10
        }
    },
    "pdk::gf180mcuC": {
        "STD_CELL_LIBRARY": "gf180mcu_fd_sc_mcu7t5v0",
        "FP_PDN_CHECK_NODES": 0,
        "FP_PDN_ENABLE_RAILS": 0,
        "RT_MAX_LAYER": "Metal4",
        "DIE_AREA": "0 0 3000 3000",
        "FP_DEF_TEMPLATE": "dir::fixed_dont_change/user_project_wrapper_gf180mcu.def",
        "PL_OPENPHYSYN_OPTIMIZATIONS": 0,
        "DIODE_INSERTION_STRATEGY": 0,
        "MAGIC_WRITE_FULL_LEF": 0
    }
}
```

d-m-bailey, 2024-12-19:
@redpanda3 The def file is used for pin placement on the top level. Probably not relevant to your design, so you can delete it.

If you are placing a hard macro, then you will need a lef file to define the pin locations to route to. If you have a gds file, you can use magic to create a lef file. I think the commands would be
```
gds read <gds_file>
load <macro_top>
select top cell
expand
lef write -pinsonly -hide
```
You can open this lef file in klayout and check that it looks appropriate.

redpanda3, 2024-12-20:
I just want to skip the synthesis step and use the netlist only.

donn, 2024-12-22:
In this version of OpenLane, there is no way to skip Synthesis (or any step for that matter).

The intended use-case, if you input a netlist and use `SYNTH_ELABORATE_ONLY`, Yosys should return your netlist flattened but otherwise unmodified.  As you've elected to skip the issue template for some reason, we can't tell if there's an actual issue here. Yosys is an integral part of the flow and the commands we run during flattening are necessary for netlists to be readable by OpenROAD.

If you for some reason need to *absolutely* skip Yosys and load the netlist directly into OpenROAD (at your own risk), try [OpenLane 2](https://github.com/efabless/openlane2), where the command would be as follows:

```sh
openlane --from OpenROAD.Floorplan -e nl=verilog/rtl/vortex/VX_core_top.mapped.v openlane/vortex/config.json
```

closed 2024-12-22 by donn