The-OpenROAD-Project/OpenLane #2178: Not able to build sky130_fd_sc_hs using the guide given at https://openlane.readthedocs.io/en/latest/usage/custom_pdk_builds.html

dhruvvyas224, 2024-12-21:
### Description

I am trying to use the `sky130_fd_sc_hs` standard cell library (SCL) and have followed the instructions provided on [this page](https://openlane.readthedocs.io/en/latest/usage/custom_pdk_builds.html). However, the PDK is not being built as expected despite following the steps.

To aid in troubleshooting, I have attached the zipped logs, including `config.log` and `install.log`.

I would greatly appreciate assistance from anyone familiar with this issue. Thank you in advance. 
[build_sky130-2024-12-20-20-53-14.zip](https://github.com/user-attachments/files/18218564/build_sky130-2024-12-20-20-53-14.zip) @donn @maliberty 

PS: I apologize for not adhering to the issue template.

### Expected Behavior

The PDK should be built with the additional SCLs.

### Environment report

```shell
.
```

### Reproduction material

.

### Relevant log output

```shell
.
```

donn, 2024-12-21:
Well, the version of Magic you're using for the build (8.3.50???) is prehistoric, so that's why.

This page of documentation is out of date: you can just `volare enable --pdk sky130 -l all` now. We pre-build all SCLs.

I will update the documentation on Sunday.

donn, 2024-12-21:
Also, please do not @ maintainers. Matt especially has nothing to do with Volare (and little to do with OpenLane) and is a very busy person. The OpenLane team scans issues and gets to them in due time.

dhruvvyas224, 2024-12-21:
> Also, please do not @ maintainers. Matt especially has nothing to do with Volare (and little to do with OpenLane) and is a very busy person. The OpenLane team scans issues and gets to them in due time.

Thank you very much @donn for the quick reply 🙏. I will keep in consideration what you have said.

dhruvvyas224, 2024-12-21:
> Well, the version of Magic you're using for the build (8.3.50???) is prehistoric, so that's why.
> 
> This page of documentation is out of date: you can just `volare enable --pdk sky130 -l all` now. We pre-build all SCLs.
> 
> I will update the documentation on Sunday.

Actually, I installed OpenLane following the tutorial at [[this link](https://openlane.readthedocs.io/en/latest/getting_started/installation/installation_ubuntu.html)](https://openlane.readthedocs.io/en/latest/getting_started/installation/installation_ubuntu.html) and have been using it successfully for various projects. The default standard cell library (`sky130_fd_sc_hd`) works fine.  

However, for a time-critical project, I needed to use the `sky130_fd_sc_hs` library, which is not built by default. Following the instructions on [[this page](https://openlane.readthedocs.io/en/latest/usage/custom_pdk_builds.html)](https://openlane.readthedocs.io/en/latest/usage/custom_pdk_builds.html), I attempted to build it but encountered errors, and the build process failed.  

It seems the version of Magic that was installed as part of the OpenLane installation (per the tutorial linked above) is outdated, which might be the root cause of the issue.  

I will try the `volare enable --pdk sky130 -l all` approach you suggested and report back. Thank you again for pointing this out and for your efforts to update the documentation!

dhruvvyas224, 2024-12-21:
> Well, the version of Magic you're using for the build (8.3.50???) is prehistoric, so that's why.
> 
> This page of documentation is out of date: you can just `volare enable --pdk sky130 -l all` now. We pre-build all SCLs.
> 
> I will update the documentation on Sunday.

Thank you for the clarification.

I installed OpenLane following the tutorial at [this link](https://openlane.readthedocs.io/en/latest/getting_started/installation/installation_ubuntu.html) and have been using it successfully for various projects. The default standard cell library (`sky130_fd_sc_hd`) works fine.

However, for a time-critical project, I needed to use the `sky130_fd_sc_hs` library, which is not built by default. Following the instructions on [this page](https://openlane.readthedocs.io/en/latest/usage/custom_pdk_builds.html), I attempted to build it but encountered errors, and the build process failed.

It seems the version of Magic that was installed as part of the OpenLane installation (per the tutorial linked above) is outdated, which might be the root cause of the issue.

I will try the `volare enable --pdk sky130 -l all` approach you suggested and report back. Thank you again for pointing this out and for your efforts to update the documentation!

donn, 2024-12-21:
The OpenLane installation does not install Magic on your computer, only in a container.

donn, 2024-12-22:
Fixed in `e73fb3c57e687a0023fcd4dcfd1566ecd478362a`: https://openlane.readthedocs.io/en/latest/usage/custom_pdk_builds.html

closed 2024-12-22 by donn