The-OpenROAD-Project/OpenLane #2192: Problem with standard OpenLane compilation (make)

alimuhtaroglu, 2025-09-10:
### Description

Hi,
The standard OpenLane installation steps worked for me before. Reinstalling on a new Ubuntu virtual machine, I run into the following problem:
Commands:
....
git clone https://github.com/The-OpenROAD-Project/OpenLane
cd OpenLane
sudo apt install python3.12-venv
make

Get the following:

make[1]: Entering directory '/home/<eda_install_directory>/OpenLane'
Error response from daemon: manifest unknown
make[1]: *** [Makefile:102: pull-openlane] Error 1
make[1]: Leaving directory '/home/<eda_install_directory>/OpenLane'
make[1]: Entering directory '/home/<eda_install_directory>OpenLane'
make[1]: *** No rule to make target 'venv/created', needed by 'openlane'.  Stop.
make[1]: Leaving directory '/home/<eda_install_directory>/OpenLane'
make: *** [Makefile:105: get-openlane] Error 2

### Expected Behavior

Make should have run smoothly, as it did before.

### Environment report

```shell
Kernel: Linux v6.6.87.2-microsoft-standard-WSL2
Distribution: ubuntu 24.04
Python: v3.12.3 (OK)
Container Engine: docker v27.5.1 (OK)
OpenLane Git Version: b729aa514c499a24467377dd9910fa78858ebadf
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: d12db87685f4c8859f9296f84e85a48707e249a0, tested: 0fe599b2afb6708d281543108caf8310912f54af)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
The version of magic used in building the PDK does not match the version OpenLane was tested on (installed: 6195c20d3d70fbfd47d75103520ada6caf7227cd, tested: 8b3bb1ae771da1673ac148987466ff31c63513cd)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

b729aa51 2025-09-06T09:35:45+03:00 docs: clarify openlane/librelane status - Mohamed Gaber -  (HEAD -> superstable, origin/superstable, origin/HEAD)
413d3010 2025-07-11T18:54:28+03:00 misc: update copyright year, podman fix - Mohamed Gaber -  (tag: 1.0.1)
e9a35339 2025-07-11T18:54:28+03:00 ci: uses cache for PDKs - Mohamed Gaber -  ()
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### Reproduction material

Just standard installation of OpenLane.

### Relevant log output

```shell
make[1]: Entering directory '/home/<eda_install_directory>/OpenLane'
Error response from daemon: manifest unknown
make[1]: *** [Makefile:102: pull-openlane] Error 1
make[1]: Leaving directory '/home/<eda_install_directory>/OpenLane'
make[1]: Entering directory '/home/<eda_install_directory>OpenLane'
make[1]: *** No rule to make target 'venv/created', needed by 'openlane'.  Stop.
make[1]: Leaving directory '/home/<eda_install_directory>/OpenLane'
make: *** [Makefile:105: get-openlane] Error 2
```

young03600, 2025-09-14:
> If you don't mind working with an older version of Openlane, run the following commands:
> 
> ```
> sudo apt-get update
> sudo apt-get upgrade
> sudo apt install -y build-essential python3 python3-venv python3-pip make git
> git clone --depth 1 https://github.com/The-OpenROAD-Project/OpenLane.git
> cd OpenLane/
> git checkout 413d301090a476f8d34cf24dc2447da17dfab187
> make
> make test
> ```

Hi,
I tried to change to the provided commit.
But I got black and flask8 error when installing the requirements.

I changed the version to:
black>=21
flake8>=7
Not sure any problem, but the installation works though.

donn, 2025-09-15:
Confirmed, working on a fix.

donn, 2025-09-15:
To keep you all looped in, it's two issues:

- The CI didn't "publish" on the latest commit to `superstable`, leaving this commit dockerless
- The fallback is to build the Docker image with `make openlane`, but that had an invalid make target.

Currently testing the fix.

closed 2025-09-15 by donn