The-OpenROAD-Project/OpenLane #2194: PDK130 No found on WSL , OpenLane Installation

XACKIES, 2025-09-24:
Hi , i am mac . i have a problem with make OpenLane on WSL 

i try to follow this section https://openlane.readthedocs.io/en/latest/getting_started/installation/installation_win.html 
but it is not work for me

i already  updated this 
```
sudo apt-get update
sudo apt-get upgrade
sudo apt install -y build-essential python3 python3-venv python3-pip make git

```

[image]

donn, 2025-09-24:
@d-m-bailey You'll see in the screenshot that it does supply the PDK, so that's not a concern.

d-m-bailey, 2025-09-24:
@donn You'll see in the screen shot error message, that the `pdk-family` needs to be specified. ;)

donn, 2025-09-24:
Missing argument is the version, not the PDK.

d-m-bailey, 2025-09-25:
@donn I stand corrected.

So either `--pdk-family` OR `--pdk` is ok. I assumed that both were required.  I appreciate the clarification.

XACKIES, 2025-09-25:
> Hiya Mac!
> 
> This means `./dependencies/tool_metadata.yml` was not found by Ciel, so can you run `ls ./dependencies/tool_metadata.yml`?

There must be something like a key, right?

[image]

XACKIES, 2025-09-25:
> [@donn](https://github.com/donn) I stand corrected.
> 
> So either `--pdk-family` OR `--pdk` is ok. I assumed that both were required. I appreciate the clarification.

Do i install sth ? 

[image]

d-m-bailey, 2025-09-25:
@XACKIES try
```
cat ./dependencies/tool_metadata.yml
```
and let us know the result.

XACKIES, 2025-09-25:
@d-m-bailey 
> and let us know the result.

Here is : 

[image]

[image]

d-m-bailey, 2025-09-25:
@XACKIES Thanks. I don't know the value of continuing to 

You might want to checkout the updated version librelane https://librelane.readthedocs.io/en/latest/index.html.

XACKIES, 2025-09-27:
Hi guys, I solved the problem by reinstalling it, so it works!! LOL XD

Thank you all, I really appreciate the community's help.